robotology/yarp #2226: Unable to compile YARP in windows

GiulioRomualdi, 2020-03-06:
When I try to compile YARP in windows (https://github.com/robotology/yarp/commit/06b62f05336b1f1462242636908787bcee5d6c53)  all the components that uses `rosmsg` fail
[image]

Is there someone that can help me? @drdanz @traversaro 

```
1>------ Build started: Project: ZERO_CHECK, Configuration: Release x64 ------
2>------ Build started: Project: YARP_priv_hmac, Configuration: Release x64 ------
3>------ Build started: Project: YARP_priv_tinyxml, Configuration: Release x64 ------
4>------ Build started: Project: YARP_priv_sqlite, Configuration: Release x64 ------
5>------ Build started: Project: yarpidl_thrift, Configuration: Release x64 ------
6>------ Build started: Project: YARP_priv_qcustomplot, Configuration: Release x64 ------
7>------ Build started: Project: YARP_priv_xmlrpcpp, Configuration: Release x64 ------
8>------ Build started: Project: yarpmod, Configuration: Release x64 ------
9>------ Build started: Project: yarpscope, Configuration: Release x64 ------
2>YARP_priv_hmac.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\lib\Release\YARP_priv_hmac.lib
10>------ Build started: Project: YARP_os, Configuration: Release x64 ------
3>YARP_priv_tinyxml.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\lib\Release\YARP_priv_tinyxml.lib
11>------ Build started: Project: yarpview, Configuration: Release x64 ------
6>Automatic MOC for target YARP_priv_qcustomplot
4>YARP_priv_sqlite.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\lib\Release\YARP_priv_sqlite.lib
12>------ Build started: Project: yarpcar, Configuration: Release x64 ------
11>Automatic MOC for target yarpview
5>yarpidl_thrift.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\bin\Release\yarpidl_thrift.exe
7>YARP_priv_xmlrpcpp.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\lib\Release\YARP_priv_xmlrpcpp.lib
12>yarpcar.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\bin\Release\yarpcar.dll
11>yarpview.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\bin\Release\yarpview.exe
9>Automatic MOC for target yarpscope
8>yarpmod.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\bin\Release\yarpmod.dll
10>YARP_os.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\bin\Release\YARP_os.dll
6>YARP_priv_qcustomplot.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\lib\Release\YARP_priv_qcustomplot.lib
9>yarpscope.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\bin\Release\yarpscope.exe
13>------ Build started: Project: YARP_init, Configuration: Release x64 ------
14>------ Build started: Project: YARP_sig, Configuration: Release x64 ------
15>------ Build started: Project: YARP_companion, Configuration: Release x64 ------
16>------ Build started: Project: YARP_name, Configuration: Release x64 ------
17>------ Build started: Project: YARP_run, Configuration: Release x64 ------
18>------ Build started: Project: yarp_human, Configuration: Release x64 ------
17>YARP_run.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\bin\Release\YARP_run.dll
18>yarp_human.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\lib\yarp\Release\yarp_human.dll
16>YARP_name.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\bin\Release\YARP_name.dll
13>YARP_init.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\bin\Release\YARP_init.dll
14>YARP_sig.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\bin\Release\YARP_sig.dll
15>YARP_companion.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\bin\Release\YARP_companion.dll
19>------ Build started: Project: yarp_portmonitor, Configuration: Release x64 ------
19>yarp_portmonitor.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\lib\yarp\Release\yarp_portmonitor.dll
20>------ Build started: Project: yarp_shmem, Configuration: Release x64 ------
20>yarp_shmem.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\lib\yarp\Release\yarp_shmem.dll
21>------ Build started: Project: yarp_xmlrpc, Configuration: Release x64 ------
22>------ Build started: Project: yarpidl_rosmsg, Configuration: Release x64 ------
23>------ Build started: Project: QtYARPScopePlugin, Configuration: Release x64 ------
24>------ Build started: Project: YARP_serversql, Configuration: Release x64 ------
22>yarpidl_rosmsg.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\bin\Release\yarpidl_rosmsg.exe
25>------ Build started: Project: YARP_math, Configuration: Release x64 ------
26>------ Build started: Project: multipleAnalogSensorsSerializations, Configuration: Release x64 ------
23>Automatic MOC for target QtYARPScopePlugin
24>YARP_serversql.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\bin\Release\YARP_serversql.dll
25>YARP_math.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\bin\Release\YARP_math.dll
27>------ Build started: Project: YARP_wire_rep_utils, Configuration: Release x64 ------
28>------ Build started: Project: joypadcontrolnetutils, Configuration: Release x64 ------
27>YARP_wire_rep_utils.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\lib\Release\YARP_wire_rep_utils.lib
28>joypadcontrolnetutils.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\src\devices\JoypadControlNetUtils\joypadcontrolnetutils.dir\Release\joypadcontrolnetutils.lib
21>yarp_xmlrpc.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\lib\yarp\Release\yarp_xmlrpc.dll
29>------ Build started: Project: YARP_rosmsg_gen_diagnostic_msgs, Configuration: Release x64 ------
29>Generating code from ../../../extern/ros/common_msgs/diagnostic_msgs/msg/DiagnosticArray.msg
29>C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(209,5): error MSB6006: "cmd.exe" exited with code -1073741515.
29>Done building project "YARP_rosmsg_gen_diagnostic_msgs.vcxproj" -- FAILED.
30>------ Build started: Project: YARP_rosmsg_gen_geometry_msgs, Configuration: Release x64 ------
31>------ Build started: Project: YARP_rosmsg_gen_native, Configuration: Release x64 ------
26>multipleAnalogSensorsSerializations.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\src\devices\multipleAnalogSensorsMsgs\multipleAnalogSensorsSerializations.dir\Release\multipleAnalogSensorsSerializations.lib
[... 1510 lines cut ...]
98>c:\users\icub\sb\robotology\yarp\src\devices\remoteframegrabber\RemoteFrameGrabber.h(593): warning C4996: 'yarp::dev::IFrameGrabberControls::getIris': Use getFeature(YARP_FEATURE_IRIS, v) instead
98>C:\Users\icub\sb\robotology\YARP\src\libYARP_dev\src\yarp/dev/FrameGrabberInterfaces.h(507): note: see declaration of 'yarp::dev::IFrameGrabberControls::getIris'
98>c:\users\icub\sb\robotology\yarp\src\devices\remoteframegrabber\RemoteFrameGrabber.h(404): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data
98>c:\users\icub\sb\robotology\yarp\src\devices\remoteframegrabber\RemoteFrameGrabber.h(405): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data
98>Generating Code...
101>Building Custom Rule C:/Users/icub/sb/robotology/YARP/src/devices/ServerSerial/CMakeLists.txt
101>CMake does not need to re-run because C:/Users/icub/sb/build/robotology/YARP/src/devices/ServerSerial/CMakeFiles/generate.stamp is up-to-date.
102>Building Custom Rule C:/Users/icub/sb/robotology/YARP/src/devices/test_grabber/CMakeLists.txt
102>CMake does not need to re-run because C:/Users/icub/sb/build/robotology/YARP/src/devices/test_grabber/CMakeFiles/generate.stamp is up-to-date.
102>yarp_plugin_yarpmod_test_grabber.cpp
99>C:\Users\icub\sb\robotology\YARP\src\libYARP_os\src\yarp/os/Port.h(248): warning C4996: 'yarp::os::Contactable::setCallbackLock': Use setCallbackLock with std::mutex instead
99>C:\Users\icub\sb\robotology\YARP\src\libYARP_os\src\yarp/os/Contactable.h(338): note: see declaration of 'yarp::os::Contactable::setCallbackLock'
93>Done building project "yarp_multipleanalogsensorsserver.vcxproj".
103>------ Build started: Project: yarp_test_motor, Configuration: Release x64 ------
99>C:\Users\icub\sb\build\robotology\YARP\src\devices\RobotDescriptionClient\yarp_plugin_yarpmod_robotDescriptionClient.cpp(33): warning C4267: 'return': conversion from 'size_t' to 'int', possible loss of data
99>RobotDescriptionClient.cpp
97>   Creating library C:/Users/icub/sb/build/robotology/YARP/lib/Release/yarp_remote_controlboard.lib and object C:/Users/icub/sb/build/robotology/YARP/lib/Release/yarp_remote_controlboard.exp
97>yarp_remote_controlboard.vcxproj -> C:\Users\icub\sb\build\robotology\YARP\lib\yarp\Release\yarp_remote_controlboard.dll
101>yarp_plugin_yarpmod_serial.cpp
[... cut ...]
92>Done building project "yarp_multipleanalogsensorsremapper.vcxproj".
100>C:\Users\icub\sb\robotology\YARP\src\libYARP_os\src\yarp/os/Port.h(248): warning C4996: 'yarp::os::Contactable::setCallbackLock': Use setCallbackLock with std::mutex instead
104>------ Build started: Project: yarp_transformClient, Configuration: Release x64 ------
102>C:\Users\icub\sb\robotology\YARP\src\libYARP_os\src\yarp/os/Port.h(248): warning C4996: 'yarp::os::Contactable::setCallbackLock': Use setCallbackLock with std::mutex instead
102>C:\Users\icub\sb\robotology\YARP\src\libYARP_o

traversaro, 2020-03-06:
The actual error is `error MSB6006: "cmd.exe" exited with code -1073741515-` . I guess that the ACE dll is not the PATH.

traversaro, 2020-03-06:
How did you installed the dependencies? 

This by the way is  quite a common failure mode on Windows, and if we could drop the linking of ACE from the rosmgs generator it would be great.

traversaro, 2020-03-06:
See the following related lines in the robotology-superbuild docs (https://github.com/robotology/robotology-superbuild#system-libraries):   

> Furthermore, as during the build process some programs that use vcpkg-install dlls are executed, you need to make sure that the vcpkg dll are in the path, by appending to the PATH the C:/robotology/vcpkg/installed/x64-linux/bin and C:/robotology/vcpkg/installed/x64-linux/bin directories.

Obviously, this changes depending on how you installed the ACE dependencies. You can also use the robotology-dependencies installer available at https://github.com/robotology/robotology-superbuild/releases/tag/v2020.02 that installs automatically the pre-compiled version of vcpkg in `C:/robotology/idjl` and also offers an option to automatically update the `PATH` variables to avoid this problem, but I still need to document its use.

GiulioRomualdi, 2020-03-06:
I'm using the vcpkg as described in https://github.com/robotology/robotology-superbuild#system-libraries

GiulioRomualdi, 2020-03-06:
@traversaro 
> Furthermore, as during the build process some programs that use vcpkg-install dlls are executed, you need to make sure that the vcpkg dll are in the path, by appending to the PATH the C:/robotology/vcpkg/installed/x64-linux/bin and C:/robotology/vcpkg/installed/x64-linux/bin directories.

That's wrong right? `C:/robotology/vcpkg/installed/x64-windows/bin` should be the corrected path. 
Furthermore `C:/robotology/vcpkg/installed/x64-linux/bin` is repeated twice. S

traversaro, 2020-03-06:
If in the future you want to debug better the MSB6006 error, check how to check the output logs of the command that actually failed: https://stackoverflow.com/questions/13118947/error-msb6006-cmd-exe-exited-with-code-1 .

traversaro, 2020-03-06:
> 
> 
> @traversaro
> 
> > Furthermore, as during the build process some programs that use vcpkg-install dlls are executed, you need to make sure that the vcpkg dll are in the path, by appending to the PATH the C:/robotology/vcpkg/installed/x64-linux/bin and C:/robotology/vcpkg/installed/x64-linux/bin directories.
> 
> That's wrong right? `C:/robotology/vcpkg/installed/x64-windows/bin` should be the corrected path.
> Furthermore `C:/robotology/vcpkg/installed/x64-linux/bin` is repeated twice. Shouldn't be `C:/robotology/vcpkg/installed/x64-windows/bin` `C:/robotology/vcpkg/installed/x64-windows/lib`?

That is correct, sorry for the typo. That is exactly why I typically ask for feedback on PRs. : )  
If you append the correct paths, it works? Could you provide a PR to fix the docs? Thanks!

traversaro, 2020-03-06:
I think I have suspect on this issue, also based on the fact that by inspecting the laptop with @GiulioRomualdi it seems that the missing dll was `vcruntime140_1.dll` (see https://devblogs.microsoft.com/cppblog/making-cpp-exception-handling-smaller-x64/ and https://gitlab.kitware.com/cmake/cmake/issues/19829). 
To confirm, what was the Visual Studio 2019 version on your machine?

GiulioRomualdi, 2020-03-09:
Actually no. The visual studio installed in the machine is Visual Studio 2017

randaz81, 2020-03-09:
Have you generated all the required messages by enabling `ALLOW_IDL_GENERATION=ON` in yarp ccmake?

traversaro, 2020-03-09:
> 
> 
> Have you generated all the required messages by enabling `ALLOW_IDL_GENERATION=ON` in yarp ccmake?

The problem was that the dependency installer just supports Visual Studio  2019, and @GiulioRomualdi was able  to get it work by installing the "Microsoft Visual C++ Redistributable for Visual Studio 2015, 2017 and 2019", that installed the missing  `vcruntime140_1.dll` (see https://stackoverflow.com/questions/58828982/vcruntime140-1-dll-missing-after-building-solution-with-c and https://devblogs.microsoft.com/cppblog/making-cpp-exception-handling-smaller-x64/). I updated the docs in https://github.com/robotology/robotology-superbuild/releases/tag/v2020.02 to  mention that Visual Studio 2019 is required, so I think we can close the issue now, what do you think @GiulioRomualdi ?

GiulioRomualdi, 2020-03-09:
Yes, we can close it, thank you very much!

closed 2020-03-09 by traversaro

Nicogene, 2020-03-10:
> to mention that Visual Studio 2019 is required, 

Just to understand, so we currently support only VS2019?

traversaro, 2020-03-10:
> 
> 
> > to mention that Visual Studio 2019 is required,
> 
> Just to understand, so we currently support only VS2019?

Officially yes, due to the fact that libraries compiled with VS2015 are compatible with executable compiled with VS2019, but not the reverse (see https://docs.microsoft.com/en-us/cpp/porting/binary-compat-2015-2017?view=vs-2019). As on GitHub actions for now only the v142 toolset (i.e. VS2019) is available, we are forced to just support that. In practice I think that as of this moment Microsoft did not break even "backward" compatibility, so using VS2017 will probably work fine (if you install the redistributable) but there is not guarantee that it will continue to work.