robotology/yarp #2233: JoypadControlServer and RGBDSensorWrapper print strange messages based on the PolyDriverList key

traversaro, 2020-03-24:
**Describe the bug**
It seems that both JoypadControlServer and RGBDSensorWrapper  prints a strange message after inspecting the PolyDriverList key, see: 
https://github.com/robotology/yarp/blob/4785668ae205f87a62c7777f77517c3b5558402c/src/devices/JoypadControlServer/JoypadControlServer.cpp#L788
and 
https://github.com/robotology/yarp/blob/8bc852eb80d463df96dff71a4be9dc2f7e01bae3/src/devices/RGBDSensorWrapper/RGBDSensorWrapper.cpp#L515 . The mesage is confusing because the user think it is an error, even if it is actually not.

**To Reproduce**
Steps to reproduce the behavior:
Run a simulated depth camera in Gazebo, and  you will get as output:
~~~
[INFO]created device <gazebo_depthCamera>. See C++ class GazeboYarpDepthCameraDriver for documentation.
[INFO]RGBDSensorWrapper: Bad name Dude!! 
~~~

**Expected behavior**
No confusing (and sexist? : ) ) message. 

**Configuration (please complete the following information):**
 - OS: Ubuntu 18.04
 - yarp version: https://github.com/robotology/yarp/commit/964bb26fa4791d83d72882711ea7509306248106
-  compiler: GCC

**Additional context**
I think that part of the code can be just deleted, but I do not know if there is any reason why it was there @drdanz @aerydna .

traversaro, 2020-03-24:
FYI @prajval10

aerydna, 2020-03-25:
i0m checking it

aerydna, 2020-03-25:
if i remember well i wrote the error (which btw is not sexist.. its as neutral as saying "hey guys" to a group of people or "mankind".. "dude" stands for "person") 
but sincerely i didn't write the check itself.. and i don't rembember why its there. 

in the case of joypadControlServer i took that part from the RGBDSensorWrapper.. maybe @barbalberto could remember why we where doing that check..

drdanz, 2020-07-14:
Fixed by 06f2b872541a9d911432d15ff480cf037fc63822 and f08275b3013e409b4f5364ac29694d35704bc841

commit f08275b3 (2020-07-14): devices/RGBDSensorWrapper: Clean log messages

Fixes #2233
closed 2020-07-14 by drdanz