robotology/yarp #2237: Question: undefined method `undent' for #<String:0x00007fa85b9dcca8>

GengTianyu, 2020-03-27:
After I installed the yarp, some info display the screen:

```
Error: undefined method `undent' for #<String:0x00007fa85b9dcca8> 
Please report this issue:
  https://docs.brew.sh/Troubleshooting
/usr/local/Homebrew/Library/Taps/robotology/homebrew-formulae/Formula/yarp.rb:72:in `caveats'
/usr/local/Homebrew/Library/Homebrew/caveats.rb:19:in `caveats'
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/forwardable.rb:224:in `empty?'
/usr/local/Homebrew/Library/Homebrew/formula_installer.rb:622:in `caveats'
/usr/local/Homebrew/Library/Homebrew/formula_installer.rb:671:in `finish'
/usr/local/Homebrew/Library/Homebrew/cmd/install.rb:330:in `install_formula'
/usr/local/Homebrew/Library/Homebrew/cmd/install.rb:261:in `block in install'
/usr/local/Homebrew/Library/Homebrew/cmd/install.rb:259:in `each'
/usr/local/Homebrew/Library/Homebrew/cmd/install.rb:259:in `install'
/usr/local/Homebrew/Library/Homebrew/brew.rb:103:in `<main>'
```

I have no idea about this, for every step have installed successfully.
Please help me, Thank you very much.

GengTianyu, 2020-03-27:
oh, I am sorry that i forget to tell the info about my os. It is macOS Catalina 10.15.4.

drdanz, 2020-04-07:
Sorry for the delay.
How did you install YARP?
Is this error printed while you install it or just after the installation? If it is during the installation, can you provide the full log?

closed 2020-07-14 by drdanz