robotology/yarp #2247: macOs builds are failing on master branch

Nicogene, 2020-04-10:
**Describe the bug**
I noticed that during the nightly build of `icub-tests` the compilation on macOs failed : https://github.com/robotology/icub-tests/runs/575776309

Actually it seem a problem with `cmath.h` header. cc @traversaro @drdanz 

Here the error:

```
 In file included from /Users/runner/runners/2.168.0/work/icub-tests/icub-tests/yarp/src/libYARP_serversql/src/yarp/serversql/Server.cpp:23:
/Applications/Xcode_11.3.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:318:9: error: no member named 'signbit' in the global namespace
using ::signbit;
      ~~^
/Applications/Xcode_11.3.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:319:9: error: no member named 'fpclassify' in the global namespace
using ::fpclassify;
      ~~^
/Applications/Xcode_11.3.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:320:9: error: no member named 'isfinite' in the global namespace; did you mean 'finite'?
using ::isfinite;
      ~~^
/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/math.h:749:12: note: 'finite' declared here
extern int finite(double)
           ^
In file included from /Users/runner/runners/2.168.0/work/icub-tests/icub-tests/yarp/src/libYARP_serversql/src/yarp/serversql/Server.cpp:23:

       ~~^
/Applications/Xcode_11.3.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:320:9: error: no member named 'isfinite' in the global namespace; did you mean 'finite'?
using ::isfinite;
      ~~^
/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/math.h:749:12: note: 'finite' declared here
extern int finite(double)
           ^
In file included from /Users/runner/runners/2.168.0/work/icub-tests/icub-tests/yarp/src/libYARP_serversql/src/yarp/serversql/Server.cpp:23:
/Applications/Xcode_11.3.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:321:9: error: no member named 'isinf' in the global namespace
using ::isinf;
      ~~^
/Applications/Xcode_11.3.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:322:9: error: no member named 'isnan' in the global namespace
using ::isnan;
      ~~^

 /Applications/Xcode_11.3.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:323:9: error: no member named 'isnormal' in the global namespace
using ::isnormal;
      ~~^
/Applications/Xcode_11.3.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:324:7: error: no member named 'isgreater' in the global namespace; did you mean '::std::greater'?
using ::isgreater;
      ^~
/Applications/Xcode_11.3.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/functional:728:29: note: '::std::greater' declared here
struct _LIBCPP_TEMPLATE_VIS greater : binary_function<_Tp, _Tp, bool>
                            ^
In file included from /Users/runner/runners/2.168.0/work/icub-tests/icub-tests/yarp/src/libYARP_serversql/src/yarp/serversql/Server.cpp:23:
/Applications/Xcode_11.3.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:325:7: error: no member named 'isgreaterequal' in the global namespace; did you mean '::std::greater_equal'?
using ::isgreaterequal;
      ^~
/Applications/Xcode_11.3.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/functional:757:29: note: '::std::greater_equal' declared here

 In file included from /Users/runner/runners/2.168.0/work/icub-tests/icub-tests/yarp/src/libYARP_serversql/src/yarp/serversql/Server.cpp:23:
/Applications/Xcode_11.3.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:326:9: error: no member named 'isless' in the global namespace
using ::isless;
      ~~^
/Applications/Xcode_11.3.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:327:9: error: no member named 'islessequal' in the global namespace
using ::islessequal;
      ~~^
/Applications/Xcode_11.3.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:328:9: error: no member named 'islessgreater' in the global namespace
using ::islessgreater;
      ~~^
/Applications/Xcode_11.3.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:329:9: error: no member named 'isunordered' in the global namespace
using ::isunordered;
      ~~^
/Applications/Xcode_11.3.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cmath:330:9: error: no member named 'isunordered' in the global namespace
using ::isunordered;
      ~~^

```

**To Reproduce**
Steps to reproduce the behavior:
Compile latest master on macOs

**Expected behavior**
It should compile.

**Configuration (please complete the following information):**
 - OS: macOs
 - yarp version: `master`
-  compiler: 

**Additional context**
Add any other context about the problem here.

Nicogene, 2020-04-10:
Ah sorry just saw https://stackoverflow.com/questions/58628377/catalina-c-using-cmath-headers-yield-error-no-member-named-signbit-in-th

traversaro, 2020-04-10:
Related superbuild issues: 
* https://github.com/robotology/robotology-superbuild/issues/375
* https://github.com/robotology/robotology-superbuild/issues/377

Nicogene, 2020-04-10:
The @traversaro 's workaround (https://github.com/robotology/robotology-superbuild/issues/375#issuecomment-611562849) semms ok.

traversaro, 2020-04-10:
If the issue is reproducible on real macOs setups, it probably make sense to do so at the YARP level, I asked to macOs users if they could reproduce it in https://github.com/robotology/robotology-superbuild/issues/375#issuecomment-611510282 but at the moment I did not received any reply from them.

closed 2020-07-14 by drdanz