robotology/yarp #2248: New "companion" commands proposal

elandini84, 2020-04-16:
Since there is'nt one yet, I'd like to propose to add two commands that would allow to set get a connection quality of service parameters:
1. qos (for package priority)
2. sched (for thread policy and priority)

### QOS command 

The command will have the following base synthax

```
$ yarp qos <mode>
```

There are three different values for `mode`
* `describe`
  ```
  $ yarp qos describe [value]
  ```
  * If value `value` is omitted a table with all the DSCP values is returned
    <br/>The following is an example of table (not at all the definitive one)
    | DSCP class | DSCP (bin) | DSCP (hex) | DSCP (dec) | ToS (dec) | ToS (hex)	 | ToS (bin) | ToS Prec. (bin) | ToS Prec. (dec) | ToS Delay Flag | ToS Throghput Flag | ToS Reliability Flag | TOS String Format | Level |
    | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- |
    | CS0 |	000000 | 0×00 | 0 | 0 | 0×00| 00000000 |000 | 0 | 0 | 0 | 0 | - | NORM |
    | CS1 |	001000 | 0×08 | 8 | 32 | 0×20| 00100000 |001 | 1 | 0 | 0 | 0 | Priority | - |
    | CS2 |	010000 | 0×10 | 16 | 64 | 0×40| 01000000 |010 | 2 | 0 | 0 | 0 | Immediate | - |
    | CS3 |	011000 | 0×18 | 24 | 96 | 0×60| 01100000 |011 | 3 | 0 | 0 | 0 | Flash | - |
    | CS4 |	100000 | 0×20 | 32 | 128 | 0×80| 10000000 |100 | 4 | 0 | 0 | 0 | FlashOverride | - |
    | CS5 |	101000 | 0×28 | 40 | 160 | 0×A0| 10100000 |101 | 5 | 0 | 0 | 0 | Critical | - |
    | CS6 |	110000 | 0×30 | 48 | 192 | 0×C0| 11000000 |110 | 6 | 0 | 0 | 0 | Internetworkcontrol | - |
    | CS7 |	111000 | 0×38 | 56 | 224 | 0×E0| 11100000 |111 | 7 | 0 | 0 | 0 | Networkcontrol | - |
    | AF11 | 001010 | 0×0A | 10 | 40 | 0×28| 00101000 |001 | 1 | 0 | 1 | 0 | Priority | LOW |
    | AF12 | 001100 | 0×0C | 12 | 48 | 0×30| 00110000 |001 | 1 | 1 | 0 | 0 | Priority | - |
    | AF13 | 001110 | 0×0E | 14 | 56 | 0×38| 00111000 |001 | 1 | 1 | 1 | 0 | Priority | - |
    | AF21 | 010010 | 0×12 | 18 | 72 | 0×48| 01001000 |010 | 2 | 0 | 1 | 0 | Immediate | - |
    | AF22 | 010100 | 0×14 | 20 | 80 | 0×50| 01010000 |010 | 2 | 1 | 0 | 0 | Immediate | - |
    | AF23 | 010110 | 0×16 | 22 | 88 | 0×58| 01011000 |010 | 2 | 1 | 1 | 0 | Immediate | - |
    | AF31 | 011010 | 0×1A | 26 | 104 | 0×68| 01101000 |011 | 3 | 0 | 1 | 0 | Flash | - |
    | AF32 | 011100 | 0×1C | 28 | 112 | 0×70| 01110000 |011 | 3 | 1 | 0 | 0 | Flash | - |
    | AF33 | 011110 | 0×1E | 30 | 120 | 0×78| 01111000 |011 | 3 | 1 | 1 | 0 | Flash | - |
    | AF41 | 100010 | 0×22 | 34 | 136 | 0×88| 10001000 |100 | 4 | 0 | 1 | 0 | FlashOverride | - |
    | AF42 | 100100 | 0×24 | 36 | 144 | 0×90| 10010000 |100 | 4 | 1 | 0 | 0 | FlashOverride | HIGH |
    | AF43 | 100110 | 0×26 | 38 | 152 | 0×98| 10011000 |100 | 4 | 1 | 1 | 0 | FlashOverride | - |
    | VA | 101100 | 0×2C | 44 | 176 | 0×B0| 10110000 |101 | 5 | 1 | 0 | 0 | Critical | CRIT |
    | EF | 101110 | 0×2E | 46 | 184 | 0×B8| 10111000 |101 | 5 | 1 | 1 | 0 | Critical | - |
    
    And this is how it should be displayed in the terminal:
    ```
    | DSCP class | DSCP (bin) | DSCP (hex) | DSCP (dec) | ToS (dec) | ToS (hex) | ToS (bin) | ToS Prec. (bin) | ToS Prec. (dec) | ToS Delay Flag | ToS Throghput Flag | ToS Reliability Flag | TOS Str. Format | Level |
    | CS0 | 000000 | 0×00 | 0 | 32 | 0×00| 00000000 |000 | 0 | 0 | 0 | 0 | - | NORM |
    | CS1 | 001000 | 0×08 | 8 | 32 | 0×20| 00100000 |001 | 1 | 0 | 0 | 0 | Priority | - |
    | CS2 | 010000 | 0×10 | 16 | 64 | 0×40| 01000000 |010 | 2 | 0 | 0 | 0 | Immediate | - |
    | CS3 | 011000 | 0×18 | 24 | 96 | 0×60| 01100000 |011 | 3 | 0 | 0 | 0 | Flash | - |
    | CS4 | 100000 | 0×20 | 32 | 128 | 0×80| 10000000 |100 | 4 | 0 | 0 | 0 | FlashOverride | - |
    | CS5 | 101000 | 0×28 | 40 | 160 | 0×A0| 10100000 |101 | 5 | 0 | 0 | 0 | Critical | - |
    | CS6 | 110000 | 0×30 | 48 | 192 | 0×C0| 11000000 |110 | 6 | 0 | 0 | 0 | Internetworkcontrol | - |
    | CS7 | 111000 | 0×38 | 56 | 224 | 0×E0| 11100000 |111 | 7 | 0 | 0 | 0 | Networkcontrol | - |
    | AF11 | 001010 | 0×0A | 10 | 40 | 0×28| 00101000 |001 | 1 | 0 | 1 | 0 | Priority | LOW |
    | AF12 | 001100 | 0×0C | 12 | 48 | 0×30| 00110000 |001 | 1 | 1 | 0 | 0 | Priority | - |
    | AF13 | 001110 | 0×0E | 14 | 56 | 0×38| 00111000 |001 | 1 | 1 | 1 | 0 | Priority | - |
    | AF21 | 010010 | 0×12 | 18 | 72 | 0×48| 01001000 |010 | 2 | 0 | 1 | 0 | Immediate | - |
    | AF22 | 010100 | 0×14 | 20 | 80 | 0×50| 01010000 |010 | 2 | 1 | 0 | 0 | Immediate | - |
    | AF23 | 010110 | 0×16 | 22 | 88 | 0×58| 01011000 |010 | 2 | 1 | 1 | 0 | Immediate | - |
    | AF31 | 011010 | 0×1A | 26 | 104 | 0×68| 01101000 |011 | 3 | 0 | 1 | 0 | Flash | - |
    | AF32 | 011100 | 0×1C | 28 | 112 | 0×70| 01110000 |011 | 3 | 1 | 0 | 0 | Flash | - |
    | AF33 | 011110 | 0×1E | 30 | 120 | 0×78| 01111000 |011 | 3 | 1 | 1 | 0 | Flash | - |
    | AF41 | 100010 | 0×22 | 34 | 136 | 0×88| 10001000 |100 | 4 | 0 | 1 | 0 | FlashOverride | - |
    | AF42 | 100100 | 0×24 | 36 | 144 | 0×90| 10010000 |100 | 4 | 1 | 0 | 0 | FlashOverride | HIGH |
    | AF43 | 100110 | 0×26 | 38 | 152 | 0×98| 10011000 |100 | 4 | 1 | 1 | 0 | FlashOverride | - |
    | VA | 101100 | 0×2C | 44 | 176 | 0×B0| 10110000 |101 | 5 | 1 | 0 | 0 | Critical | CRIT |
    | EF | 101110 | 0×2E | 46 | 184 | 0×B8| 10111000 |101 | 5 | 1 | 1 | 0 | Critical | - |
    ```
[... 141 lines cut ...]
    ```
    Again, the `policy` value will be returned as an integer
    It will also be possible to specify the data that you want to get (`--policy` for the policy value as an integer, `--policy-label` for the policy value as a label, or `--priority`).
    <br/>E.g.
    ```
    $ yarp sched get /w /r --src --priority
    ```
    And get
    ```
    (sched ((priority 1)))
    ```
* `set` allows to set one of the two scheduling related values
  ```
  $ yarp sched get <source port> <destination port> <port> <value>
  ```
  Differently from the `get` option, the port we want to refer to (`port` option) has to be provided.
  <br/>The `value` option has to be composed by an identification suffix (`PRT` for `priority` or `POL` for `policy`) 
  <br/>The `policy` value can be provided as a string o as an integer
  <br/> For both the scheduling parameters, if the provided value is not contained in the `describe` table or if it conflicts with the other parameter stored value, an error will be displayed
  <br/>For the sake of clarity, let's imagine a scenario where we have the `policy` value for the `src` port set to `SCHED_RR`.
  <br/>If we try to set the `priority` to 0 we would receive a message similar to this
  ```
  $ yarp sched set /w /r --src POL:0
  $ ERROR: The provided value is outside the SCHED_RR priority allowed range (1-99)
  ```

lornat75, 2020-04-16:
thanks @elandini84 

I was wondering if it could be useful to group all qos-related configurations under the same umbrella, like for example `yarp priority qos/yarp priority sched...` or `yarp channel qos/yarp channel sched` (just throwing ideas).

Besides these parameters for controlling the scheduling and TOS, in http://arxiv.org/abs/1504.02128 we also experimented with the packet scheduling policy in the linux kernel, this was useful to handle conflicts among packets received on the same host from different channels. We will probably need a different command for changing that.

Finally, to simplify a bit how to set these parameters we proposed four classes which provide a fixed subset of combinations (Table 1 in the paper above), it may be useful to provide these classes as a separate command.

drdanz, 2020-04-16:
@lornat75 The 4 classes mentioned in the paper, this is already planned, hey are the ones in the "Level" column of the table, and both the `set` and `get` subcommands will be able to handle it:

> LEVEL value: as for the DSCP alternative, the user can specify a decimale value (with the LEVEL: prefix) or the level label (with or without the suffix)

> return-as is specified (--tos, --dscp, or --level).

---

The reason why we divided `yarp qos` and `yarp sched` instead of having a single subcommand is because we came to the conclusion that while setting them at the same time might have sense from the code, from the command the syntax would be a bit confusing, since the things to handle are a lot, hence the 2 separate commands...
Changing the names of the subcommands from to `yarp priority qos` and `yarp priority sched`, might be a good idea, but this will complicate a bit the handling of the command line, it will complicate the auto-completion script, and will just add a few characters to what the user will have to write, without giving a real advantage, therefore I am not completely sure that this is worth. Anyway, as far as it is just the "name" of the subcommand, and we do not allow to set both at the same time, they will both work for me, what do other think?

drdanz, 2020-04-16:
I took the liberty to re-format the table as it should be printed on the command line
This is only an utility for the user, therefore having it printed as much readable as possible is probably a good idea

```
| DSCP  | DSCP   | DSCP  | DSCP  | TOS   | ToS   | ToS      | ToS   | ToS   | ToS   | ToS       | ToS         | ToS                 | YARP  |
| class | (bin)  | (hex) | (dec) | (dec) | (hex) | (bin)    | Prec. | Prec. | Delay | Throghput | Reliability | Str.                | Level |
|       |        |       |       |       |       |          | (bin) | (dec) | Flag  | Flag      | Flag        | Format              |       |
|-------|--------|-------|-------|-------|-------|----------|-------|-------|-------|-----------|-------------|---------------------|-------|
| CS0   | 000000 | 0×00  | 0     | 0     | 0×00  | 00000000 | 000   | 0     | 0     | 0         | 0           | -                   | NORM  |
| CS1   | 001000 | 0×08  | 8     | 32    | 0×20  | 00100000 | 001   | 1     | 0     | 0         | 0           | Priority            | -     |
| CS2   | 010000 | 0×10  | 16    | 64    | 0×40  | 01000000 | 010   | 2     | 0     | 0         | 0           | Immediate           | -     |
| CS3   | 011000 | 0×18  | 24    | 96    | 0×60  | 01100000 | 011   | 3     | 0     | 0         | 0           | Flash               | -     |
| CS4   | 100000 | 0×20  | 32    | 128   | 0×80  | 10000000 | 100   | 4     | 0     | 0         | 0           | FlashOverride       | -     |
| CS5   | 101000 | 0×28  | 40    | 160   | 0×A0  | 10100000 | 101   | 5     | 0     | 0         | 0           | Critical            | -     |
| CS6   | 110000 | 0×30  | 48    | 192   | 0×C0  | 11000000 | 110   | 6     | 0     | 0         | 0           | Internetworkcontrol | -     |
| CS7   | 111000 | 0×38  | 56    | 224   | 0×E0  | 11100000 | 111   | 7     | 0     | 0         | 0           | Networkcontrol      | -     |
| AF11  | 001010 | 0×0A  | 10    | 40    | 0×28  | 00101000 | 001   | 1     | 0     | 1         | 0           | Priority            | LOW   |
| AF12  | 001100 | 0×0C  | 12    | 48    | 0×30  | 00110000 | 001   | 1     | 1     | 0         | 0           | Priority            | -     |
| AF13  | 001110 | 0×0E  | 14    | 56    | 0×38  | 00111000 | 001   | 1     | 1     | 1         | 0           | Priority            | -     |
| AF21  | 010010 | 0×12  | 18    | 72    | 0×48  | 01001000 | 010   | 2     | 0     | 1         | 0           | Immediate           | -     |
| AF22  | 010100 | 0×14  | 20    | 80    | 0×50  | 01010000 | 010   | 2     | 1     | 0         | 0           | Immediate           | -     |
| AF23  | 010110 | 0×16  | 22    | 88    | 0×58  | 01011000 | 010   | 2     | 1     | 1         | 0           | Immediate           | -     |
| AF31  | 011010 | 0×1A  | 26    | 104   | 0×68  | 01101000 | 011   | 3     | 0     | 1         | 0           | Flash               | -     |
| AF32  | 011100 | 0×1C  | 28    | 112   | 0×70  | 01110000 | 011   | 3     | 1     | 0         | 0           | Flash               | -     |
| AF33  | 011110 | 0×1E  | 30    | 120   | 0×78  | 01111000 | 011   | 3     | 1     | 1         | 0           | Flash               | -     |
| AF41  | 100010 | 0×22  | 34    | 136   | 0×88  | 10001000 | 100   | 4     | 0     | 1         | 0           | FlashOverride       | -     |
| AF42  | 100100 | 0×24  | 36    | 144   | 0×90  | 10010000 | 100   | 4     | 1     | 0         | 0           | FlashOverride       | HIGH  |
| AF43  | 100110 | 0×26  | 38    | 152   | 0×98  | 10011000 | 100   | 4     | 1     | 1         | 0           | FlashOverride       | -     |
| VA    | 101100 | 0×2C  | 44    | 176   | 0×B0  | 10110000 | 101   | 5     | 1     | 0         | 0           | Critical            | CRIT  |
| EF    | 101110 | 0×2E  | 46    | 184   | 0×B8  | 10111000 | 101   | 5     | 1     | 1         | 0           | Critical            | -     |
```

Also `TOS (dec)` for `CS0` class was wrong (32 instead of 0), I fixed it in the table above

elandini84, 2020-04-16:
@drdanz thanks, that look way better :blush:
Thanks also for the correction. I edited the issue text.

lornat75, 2020-04-16:
For the subcommand, I understand it complicates, my suggestion was in the direction of making the commands `qos` and `sched` more specific (`sched` is very generic and may refer to other things), it can be fixed using more specific names as in `priority-qos` `priority-sched`. It is a minor comment however.

lornat75, 2020-04-16:
I am not sure I understand how one can modify the packet scheduling policy separately from the TOS. From the current proposal you seem to provide ways to specify TOS directly, or the LEVEL which affects both TOS and the packet scheduling policy.  In this way there is no way to change the packet scheduling policy without changing TOS. Am I missing something?

closed 2020-07-06 by drdanz