robotology/yarp #2262: YARP ipv6 support

plinioMoreno, 2020-04-29:
## Brief description
I wonder if YARP supports ipv6 addresses. If that is not the case, are you thinking on adding this feature?

## Detailed context
I want to setup a remote yarp server with ipv6 address, and command the robot in simulation from other ipv6 address 

## Accompanying material

drdanz, 2020-04-29:
ipv6 is partially supported in some places inside YARP (I started porting ipv4 functions to the ip4/6 version some times ago), but It was never tested, and I doubt it will work out of the box right now...
We don't have plans for adding ipv6 support any time soon, but if someone would like to work on that he is more than welcome to do it!

jwillemsen, 2020-10-07:
When using ACE you do need to enable ipv6 support when compiling ACE

drdanz, 2020-10-07:
I have been testing IPv6 recently, it's not fully working yet, but I'll probably be landing a few patches in the next few weeks.

jwillemsen, 2020-10-07:
In order to use ipv6 with ACE you should add `ipv6=1` to the `ACE_wrappers/bin/MakeProjectCreator/config/default.features` file and generate the makefiles/solutions using MPC to make sure the required compiler flags and linker settings are used.

drdanz, 2020-10-07:
Thanks @jwillemsen!
I've been testing it on debian only at the moment, and ipv6 is already enabled there: https://salsa.debian.org/debian/ace/-/blob/master/debian/default.features#L6
I'll take this into account when I start testing on windows.