robotology/yarp #2279: Incompatibily between Debug and Release builds after the introduction of the Log components

Nicogene, 2020-05-15:
**Describe the bug**
Having `yarp` compiled in `Debug` and `icub-main` (or other else repo that uses `yarp`) in `Release` (and viceversa) triggers a linking issue on `yarp::os::Log::defaultLogComponent()`

**To Reproduce**
Build and install `yarp` in `Debug`, build `icub-main` in release.

**Expected behavior**
The compatibility between Debug and Release build should be maintained.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Configuration (please complete the following information):**
 - OS: Kubuntu 20.04
 - yarp version: latest `master` - commit 3f5cee049102bde3f4cef91fa047ef8c44c9a6af
-  compiler:  `gcc (Ubuntu 9.3.0-10ubuntu2) 9.3.0`

**Additional context**
Here is the log of the failure
~~~
[ 32%] Linking CXX executable ../../../bin/emotionInterface
/usr/bin/ld: CMakeFiles/emotionInterface.dir/emotionInterfaceModule.cpp.o: in function `getHexCode[abi:cxx11](yarp::os::Bottle const&)':
emotionInterfaceModule.cpp:(.text+0x2853): undefined reference to `yarp::os::Log::defaultLogComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x287b): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x2c40): undefined reference to `yarp::os::Log::logInternalComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x2c68): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x2c97): undefined reference to `yarp::os::Log::logInternalComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x2cbf): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x2d38): undefined reference to `yarp::os::Log::do_log(yarp::os::Log::LogType, char const*, char const*, unsigned int, char const*, double, double, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x2ee0): undefined reference to `yarp_print_trace(_IO_FILE*, char const*, unsigned int)'
/usr/bin/ld: CMakeFiles/emotionInterface.dir/emotionInterfaceModule.cpp.o: in function `iCub::contrib::EmotionInterfaceModule::setColor(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
emotionInterfaceModule.cpp:(.text+0x32a1): undefined reference to `yarp::os::Log::defaultLogComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x32c9): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x341c): undefined reference to `yarp::os::Log::logInternalComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x3444): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x34ae): undefined reference to `yarp::os::Log::do_log(yarp::os::Log::LogType, char const*, char const*, unsigned int, char const*, double, double, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x35d6): undefined reference to `yarp::os::Log::logInternalComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x35fe): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x3655): undefined reference to `yarp_print_trace(_IO_FILE*, char const*, unsigned int)'
/usr/bin/ld: CMakeFiles/emotionInterface.dir/emotionInterfaceModule.cpp.o: in function `iCub::contrib::EmotionInterfaceModule::setMask(yarp::os::Bottle const&)':
emotionInterfaceModule.cpp:(.text+0x3715): undefined reference to `yarp::os::Log::defaultLogComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x373b): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x386e): undefined reference to `yarp::os::Log::defaultLogComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x3894): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x3911): undefined reference to `yarp::os::Log::defaultLogComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x3937): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x3a13): undefined reference to `yarp::os::Log::logInternalComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x3a3b): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x3aab): undefined reference to `yarp::os::Log::do_log(yarp::os::Log::LogType, char const*, char const*, unsigned int, char const*, double, double, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x3bf1): undefined reference to `yarp::os::Log::logInternalComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x3c19): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x3c89): undefined reference to `yarp::os::Log::do_log(yarp::os::Log::LogType, char const*, char const*, unsigned int, char const*, double, double, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x4216): undefined reference to `yarp::os::Log::logInternalComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x4241): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x42b4): undefined reference to `yarp::os::Log::do_log(yarp::os::Log::LogType, char const*, char const*, unsigned int, char const*, double, double, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x4401): undefined reference to `yarp::os::Log::logInternalComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x4429): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x44ce): undefined reference to `yarp::os::Log::defaultLogComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x44f4): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x4648): undefined reference to `yarp::os::Log::logInternalComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x4673): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x46b7): undefined reference to `yarp::os::Log::logInternalComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x46df): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x4f75): undefined reference to `yarp::os::Log::logInternalComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x4fa0): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x5013): undefined reference to `yarp::os::Log::do_log(yarp::os::Log::LogType, char const*, char const*, unsigned int, char const*, double, double, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x5177): undefined reference to `yarp::os::Log::logInternalComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x51a2): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x51fd): undefined reference to `yarp_print_trace(_IO_FILE*, char const*, unsigned int)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text+0x523e): undefined reference to `yarp_print_trace(_IO_FILE*, char const*, unsigned int)'
[... 20 lines cut ...]
[... cut ...]
/usr/bin/ld: emotionInterfaceModule.cpp:(.text._ZN4yarp2os9LogStreamD2Ev[_ZN4yarp2os9LogStreamD5Ev]+0x14d): undefined reference to `yarp::os::Log::do_log(yarp::os::Log::LogType, char const*, char const*, unsigned int, char const*, double, double, yarp::os::LogComponent const&)'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text._ZN4yarp2os9LogStreamD2Ev[_ZN4yarp2os9LogStreamD5Ev]+0x261): undefined reference to `yarp::os::Log::logInternalComponent()'
/usr/bin/ld: emotionInterfaceModule.cpp:(.text._ZN4yarp2os9LogStreamD2Ev[_ZN4yarp2os9LogStreamD5Ev]+0x28a): undefined reference to `yarp::os::Log::Log(char const*, unsigned int, char const*, yarp::os::LogComponent const&)'

~~~

drdanz, 2020-07-15:
I tried in several configurations to reproduce this, but I could not reproduce this issue.
@Nicogene Are you sure that it is not a setup issue?
Please remove all the old YARP libraries and CMake files and try again...

Nicogene, 2020-07-15:
I confirm that I have no more this issue, probably was related to my setup sry.

closed 2020-07-15 by Nicogene