robotology/yarp #2280: Refactoring in progress: all lidar devices will derive from Lidar2DDeviceBase

randaz81, 2020-05-16:
This issue keeps track of several PRs which have the common goal to make all lidar devices deriving from yarp class `Lidar2DDeviceBase`. This will allow to remove duplicate (and sometimes bugged/inconsistent) code in the configuration parser and limits checker.

The following components, belonging to different repos, need to be tested:

**yarp base class:**
- [ ] Lidar2DDeviceBase https://github.com/robotology/yarp/pull/2205 (in progress)

**yarp virtual sensors:**
- [ ] fakeLaser
- [ ] laserFromDepth
- [ ] laserFromExternalPort
- [ ] laserFromPointCloud

**yarp physical sensors:**
- [ ] laserHokuyo
- [ ] rpLidar (osbolete)
- [ ] rpLidar2

**cer:**
- [ ] cerDoubleLidar

**gazebo-yarp-plugins:**
- [ ] laserSensor
- [ ] doublelaser

drdanz, 2020-06-11:
@randaz81 Is this task still in progress or can we close this issue?

randaz81, 2023-10-25:
Task is now complete

closed 2023-10-25 by randaz81