robotology/yarp #2284: Error while building yarp through robotology-superbuild on Ubuntu 18.04

bemilio, 2020-05-25:
**Describe the bug**
The build of YARP fails while running `make` to build the whole [robotology-superbuild](https://github.com/robotology/robotology-superbuild) giving the following error:

```
/home/emilio/Repositories/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/impl/Storable.cpp: In function ‘std::__cxx11::string {anonymous}::fp_to_string(T)’:
/home/emilio/Repositories/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/impl/Storable.cpp:71:78: error: expected primary-expression before ‘)’ token
 #define YARP_DOUBLE_TO_STRING_MAX_LENGTH (16 + DECIMAL_DIG + YARP_DBL_EXP_DIG)
                                                                              ^
/home/emilio/Repositories/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/impl/Storable.cpp:81:14: note: in expansion of macro ‘YARP_DOUBLE_TO_STRING_MAX_LENGTH’
     char buf[YARP_DOUBLE_TO_STRING_MAX_LENGTH]; // -> see comment at the top of the file
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/emilio/Repositories/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/impl/Storable.cpp:82:19: error: ‘buf’ was not declared in this scope
     std::snprintf(buf, YARP_DOUBLE_TO_STRING_MAX_LENGTH, "%.*g", DECIMAL_DIG, x);
                   ^~~
/home/emilio/Repositories/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/impl/Storable.cpp:71:78: error: expected primary-expression before ‘)’ token
 #define YARP_DOUBLE_TO_STRING_MAX_LENGTH (16 + DECIMAL_DIG + YARP_DBL_EXP_DIG)
                                                                              ^
/home/emilio/Repositories/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/impl/Storable.cpp:82:24: note: in expansion of macro ‘YARP_DOUBLE_TO_STRING_MAX_LENGTH’
     std::snprintf(buf, YARP_DOUBLE_TO_STRING_MAX_LENGTH, "%.*g", DECIMAL_DIG, x);

```

**To Reproduce**
follow the build instruction of [robotology-superbuild](https://github.com/robotology/robotology-superbuild) 

**Expected behavior**
A successful build

**Screenshots**
--

**Configuration (please complete the following information):**
 - OS: Kubuntu 18.04
 - yarp version: 3.3.103
-  compiler: gcc

**Additional context**
--

traversaro, 2020-05-25:
Which value do you have for your `ROBOTOLOGY_PROJECT_TAGS` option? Are you sure you don't already have a YARP installed in your system?

bemilio, 2020-05-25:
in the `CMakeCache`:
```
//The tags to be used for the robotology projects: Stable, Unstable
// or Custom. This can be changed only before the first configuration.
ROBOTOLOGY_PROJECT_TAGS:STRING=Stable

//If ROBOTOLOGY_PROJECT_TAGS is custom, this file will be loaded
// to specify the tags of the projects to use.
ROBOTOLOGY_PROJECT_TAGS_CUSTOM_FILE:FILEPATH=
```

absolutely sure, i just reinstalled linux

traversaro, 2020-05-25:
Did you manually changed the YARP version? The error that you indicated and the version that you reported seems to be from the latest master, but the branch of the `Stable` tags is `yarp-3.3`, see https://github.com/robotology/robotology-superbuild/blob/master/cmake/ProjectsTagsStable.cmake and https://github.com/robotology/robotology-superbuild/blob/master/doc/change-project-tags.md .

bemilio, 2020-05-25:
so, I first had the problem with the standard version shipped with the superbuild (which I guess is yarp-3.3). Then I tried to do `git checkout` on other branches to see if that would change anything. The exact same error was appearing using older versions such as `yarp-3.0` . After this test, I reverted to master (because I thought that was the original branch) and then I wrote this issue using the version that was appearing there.

TL;DR - the same error appears on different versions, including the stable one on branch `yarp-3.3`

traversaro, 2020-05-26:
> TL;DR - the same error appears on different versions, including the stable one on branch `yarp-3.3`

Are you sure? The error message that you report is about the `YARP_DOUBLE_TO_STRING_MAX_LENGTH` define in the file https://github.com/robotology/yarp/blob/89563716ea998f66a1722875db1564642cb346f3/src/libYARP_os/src/yarp/os/NetType.cpp, but that variable is not present in that file in the `yarp-3.3` branch: https://github.com/robotology/yarp/blob/yarp-3.3/src/libYARP_os/src/yarp/os/NetType.cpp . 

In general it is much easier to provide support to you if you try to report issues regarding problems that occur by following the build instructions in a clean  environment, not the error that you get after trying before some workarounds, as those workaround may complicate a lot the analysis of the situation. More in particular, the robotology-superbuild compilation is extensively in CI on Ubuntu 18.04 and by most users, so if something is behaving different I would try to understand what is different in your setup.

bemilio, 2020-05-26:
I don't know what happened but I am sure i checked out the `yarp-3.3` branch.

Anyway, i re-cloned the superbuild and the problem disappeared

traversaro, 2020-05-26:
Ok, I think we can close the issue then, right?

bemilio, 2020-05-26:
yes, closing

closed 2020-05-26 by bemilio