robotology/yarp #2287: yarpidl_rosmsg String Gives Failed to open /yarp/rosmsg/String.h for writing error

samialperen, 2020-06-04:
I am following the tutorial provided in [this link](https://www.yarp.it/yarp_with_ros_writing_code_topics.html). However, when I run 
` yarpidl_rosmsg String` 
It gives the following error
` Failed to open yarp/rosmsg/String.h for writing `

The funny thing is it actually creates required String.h and String.msg file into the directory that I execute the command from. What am I missing?

randaz81, 2020-06-05:
Hi @samialperen 
Which yarp version are you using? On which OS?

samialperen, 2020-06-05:
Hi @randaz81 
YARP version 3.3.103+20200515.2+gitd74a0636e
OS: Ubuntu 18.04

drdanz, 2020-06-11:
The tutorial is outdated, you should just use

```c++
#include <yarp/rosmsg/std_msgs/String.h>
```

and link to the `YARP::YARP_rosmsg_std_msgs` library (or `YARP::YARP_rosmsg` to include all ros libraries) from CMake

samialperen, 2020-06-19:
@drdanz Thank you very much!  Unfortunately, still it doesn't work. I linked to YARP_rosmsg as well. 

Here is my CMakeLists.txt
```cmake_minimum_required(VERSION 3.12)
find_package(YARP COMPONENTS os REQUIRED)
add_executable(yarp_rostopic_listener)
target_sources(yarp_rostopic_listener PRIVATE yarp_rostopic_listener.cpp)
target_link_libraries(yarp_rostopic_listener PRIVATE YARP::YARP_os YARP::YARP_init YARP::YARP_rosmsg)
```
When I try to cmake, I have the following error:
```
-- Found YARP: /usr/local/src/robot/yarp/build (found version "3.3.103+20200515.2+gitd74a0636e")
-- Configuring done
CMake Error at CMakeLists.txt:5 (add_executable):
  Target "yarp_rostopic_listener" links to target "YARP::YARP_rosmsg" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
```

drdanz, 2020-06-22:
@samialperen You need to search for the `rosmsg` component from CMake, try changing the first line of your `CMakeLists.txt` with this:

```cmake
find_package(YARP COMPONENTS os rosmsg REQUIRED)
```

samialperen, 2020-06-23:
@drdanz Thank you very much! It works now. Where can I look further to learn more about ros and yarp connection? Is there any updated tutorial?

drdanz, 2020-06-25:
@samialperen You can find the other tutorials here: https://www.yarp.it/yarp_with_ros.html
I'm sorry, some documentation is a bit outdated (at least for CMake, code generation, and "#include" parts), but the actual usage shouldn't be changed at all...

drdanz, 2020-06-25:
I'm closing this issue, feel free to re-open this (or a new one) if you have any further question

closed 2020-06-25 by drdanz