robotology/yarp #2289: segmentation fault on using yarpcar_h264

ddetommaso, 2020-06-08:
**Describe the bug**
The yarpcar_h264 plugin seems not working properly.

**To Reproduce**
Following instructions at [https://www.yarp.it/carrier_h264_howto.html](url)
I got segmentation fault typing the following command:

`yarp connect /gst /yarpview/img:i h264`

I checked and both the gstreamer server and client work correctly.

**Configuration (please complete the following information):**
 - OS: Ubuntu 18.04/20.04
 - yarp version: v3.3.2

ddetommaso, 2020-06-18:
Hi guys,

in case you still couldn't fixed this or reproduced the issue, can you post here a version of YARP that works with yarpcar_h264  ?

Thanks

Nicogene, 2020-06-18:
@valegagge is the main author of that carrier and unfortunately she will be not available for months. 
I have no experience with that carrier maybe some other @robotology/yarp-developers with more experience can handle this issue.

Sorry for the troubles

randaz81, 2024-08-30:
`h264 `carrier replaced by `gstreamer `carrier: https://github.com/robotology/yarp/pull/3127. See updated documentation.

closed 2024-08-30 by randaz81