robotology/yarp #2297: Communication issue between Windows and Linux machines

aikolina, 2020-06-22:
**Describe the bug**
We have two computers on the same network (a workstation with _Windows_ and a laptop with _Linux_), and they cannot communicate with each other through YARP.

**To Reproduce**
Steps to reproduce the behavior:

- First Test
  1. Run a `yarpserver` on Windows
  2. Run `yarp detect` on Linux --> Linux was not able to detect the yarp server! :sweat:
- Second Test
  1. Run a `yarp server` on Linux
  2. `yarp detect` on Windows --> Windows detects successfully the yarpserver! :tada:
  3. Run the `yarp-vicon-bridge`  on Windows
  4. `yarp read ... /transformServer/transforms:o` on Linux --> Linux couldn't read anything from the port

**N.B.** We have checked the yarp configurations and they are set correctly for all these tests.

After these two tests, we checked that the two computers could ping each other:

1. Windows could ping Linux
2. Linux could not ping Windows (oh, no! :cry:)

We tried to disable the Firewall on Windows, but we got the same behaviors as described above.

Then, we followed [these instruction](https://superuser.com/questions/406958/how-can-one-machine-ping-another-but-the-reverse-ping-doesnt-work) and tried enabling the Inbound rule in advanced firewall settings for ICMP (File Sharing and Printer, private profile). This allowed us to ping back and forth between the Linux and Windows machines. (We tried Experiments first and second described before, and yet not working!)

After this, we also tried to `ssh` from Linux to Windows, which was not possible because of refused connection over port 22. We tried to add another inbound rule in the advanced firewall settings to allow connections through port 22. Even by doing so, we were still not able to `ssh` from Linux to Windows.

However, in any case, the problems seen in both first and second tests were still there.

**Expected behavior**
Correct communication between Windows and Linux machines (e.g. `yarp detect` and `yarp read` nicely working).

**Configuration (please complete the following information):**
 - OS: Windows + Linux
 - yarp version: X.Y on Windows (to be checked) + YARP v. 3.3 on Linux
-  compiler: to be checked on WIndows + gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)

**Additional context**
The Windows computer we are working on is the Vicon workstation is S. Quirico.

aikolina, 2020-06-22:
Does anyone have any suggestion to handle this type of communication issue?

@kouroshD , @prashanthr05  and I spent a few hours trying to figure out what's going wrong but in vain. Any help would be appreciated. Thanks in advance!

randaz81, 2020-06-22:
Can you please check if the machine has multiple network interfaces and you are using the correct ip address?  You should be able to ping bidirectionally, before proceeding further...

aikolina, 2020-06-22:
> This allowed us to ping back and forth between the Linux and Windows machines.

We actually managed to ping in both directions! Still `yarp detect` is not working.

drdanz, 2020-06-22:
Why do you have 2 yarp server running?

aikolina, 2020-06-22:
@drdanz No, we didn't have two yarp servers running in parallel. We did **two separate tests** as described above:
> - First Test
>  1. Run a `yarpserver` on Windows
>  2. Run `yarp detect` on Linux --> Linux was not able to detect the yarp server! :sweat:
> - Second Test
>  1. Run a `yarp server` on Linux
>  2. `yarp [...]

prashanthr05, 2020-06-22:
> Do you have 2 yarp server running?

Nope.

Besides that, I did the following tests,

## yarpserver running on Linux

Being able to ping bidirectionally between windows and linux machines, both running on the same subnet.

1. `yarp conf` with IP address of my laptop connected to local network  

2. `yarpserver` 

3. `yarp read /read` on Linux and `yarp write /write /read` on windows results in creating the respective ports and throws following error on Windows ,`Failure: Do not know how to connect to tcp://read` 

   | Windows                                                      | Linux                                                        |
   | ------------------------------------------------------------ | ------------------------------------------------------------ |
   |[image]| [image]|

   

4. `yarp read /read` on Windows and `yarp write /write /read` on Linux, there is no error, but whatever I write in Linux is not received at the /read port opened in Windows

   | Windows                                                      | Linux                                                        |
   | ------------------------------------------------------------ | ------------------------------------------------------------ |
   | [image]| [image]|

These tests I did with firewall settings enabled and disabled. Same behavior.

## yarpserver running on Windows

yarpserver running on windows, unable to detect on Linux (firewall enabled/disabled)

| Windows                                                      | Linux                                                        |
| ------------------------------------------------------------ | ------------------------------------------------------------ |
| [image]| [image]|

jgvictores, 2020-06-22:
Note: regarding `yarp detect`,  I cannot find a reference, but a mantra like "relies on multicast, which may not be available on all networks" resonates (note that refers to the network, not the compiled carrier). So don't get extremely worried about it.

I'd follow these steps:
1. Reboot both machines
1. On the Windows machine, disable all firewalls. Absolutely all. Wreak havoc by making making all options from `Windows Defender Firewall with Advanced Security` (where `Windows Defender Firewall Properties` also can hide options) - as well as any antivirus - panic with lots of red semaphores (`cryptobotics` is sooo 2015 ^^)
1. On the Windows machine: `yarp conf xx.xxx.x.41 10000` (replace each `x` as a human to refer to the Windows IP)
1. On the Linux machine: `yarp conf xx.xxx.x.41 10000` (replace each `x` as a human to refer to the Windows IP)
1. On the Windows machine: `yarp server`
1. If it's not too much to ask, report on same write/read experiments in each direction. : )

PeterBowman, 2020-06-22:
Back in Ubuntu Xenial's times, I often needed to disable Linux firewall altogether by typing:

```
sudo iptables -P INPUT ACCEPT
sudo iptables -P OUTPUT ACCEPT
sudo iptables -P FORWARD ACCEPT
sudo iptables -F
```

Interestingly, Windows firewalls didn't bother me so much as Linux's. Note things may have changed on Bionic ([ref](https://linuxconfig.org/how-to-enable-disable-firewall-on-ubuntu-18-04-bionic-beaver-linux)).

jgvictores, 2020-06-22:
PS: Also, to keep things simple, disable network interfaces that are not specific for the write/read experiment (this may be done via GUI on Windows, and via `sudo ifconfig xxxxxxx down` or some `network-manager` stuff on Linux). Once things are working, there may be certain configurations you must discover in order to maintain functionality with multiple interfaces (e.g. in Ubuntu I've had to select the "use connection only for resources" option for one of the interfaces).

aikolina, 2020-06-22:
@mbrunettini If you agree, @prashanthr05 and I would like to try the suggestions we got from @PeterBowman and @jgvictores on the Vicon workstation.

mbrunettini, 2020-06-22:
1) Linux firewall is not enabled by default, so suggestion of @PeterBowman is not needed
2) if you disable network interfaces as requested by @jgvictores **VICON STOPS WORKING** and the suggestion is not resolutive

sorry, @aikolina please no

mbrunettini, 2020-06-22:
my suyggestion is to test a YARP port connection in a separate environment (outside VICON, eg. your laptop) 
- one port on WINDOWS
- one port on UBUNTU in WSL

try a connection and see if it works, like [here ](http://wiki.icub.org/wiki/Check_your_installation) 

you can also test as suggested [here](https://www.yarp.it/yarp_cluster.html)

aikolina, 2020-06-22:
Thank you very much for the suggenstions, @mbrunettini!

aikolina, 2020-06-22:
I'm closing the issue because I believe that this is a problem of a specific setup (the Vicon workstation in S. Quirico, IIT). I will open an internal issue in a more appropriate place.

@drdanz I'm not sure how it works in general but maybe it's worth deleting this issue (is it possible?) or put a kind of big "NONO" label so that people don't get confused by what it has been discussed here.

Thank you very much for all the comments anyway! :smile_cat:

closed 2020-06-22 by aikolina