robotology/yarp #2308: [Discussion] test_grabber should be fakeGrabber

drdanz, 2020-07-08:
All the fake devices, with the exception of `test_grabber` have a name that starts with `fake`.

```
fakeAnalogSensor
fakeBattery
fakeDepthCamera
fakeIMU
fakeLaser
fakeLocalizerDevice
fakeMicrophone
fakeMotionControl
fakeNavigationDevice
fakeSpeaker
fakebot
```

I noticed that the name `test_grabber` is confusing (it appears to be something related to tests), especially for newcomers, I suggest to rename it to `fakeGrabber`.

Since this device is quite used, I suggest to proceed in 3 step
1) [ASAP] Add a `fakeGrabber` device and add a warning to `test_grabber`, update documentation, and start telling everyone to use `fakeGrabber` instead of `test_grabber`
2) [In a few months] Make `test_grabber` a `DeprecatedDevice` (`allow-deprecated-devices` must be passed in order to use the device)
3) [In a major release in the future] Remove `test_grabber`

What do you think? @robotology/yarp-developers

traversaro, 2020-07-08:
I agree.

drdanz, 2020-07-08:
Same arguments are valid for `TestMotor`, even though it is not used as much as `test_grabber`

drdanz, 2020-07-27:
Actually the name should probably be `fakeFrameGrabber`, not `frameGrabber`, updating the subject...

drdanz, 2020-07-27:
Perhaps also `fakebot` should be `fakeBot` for coherence?

randaz81, 2020-07-27:
I'm not event sure if fakeBot is useful at all. 
Its functionality can be replaced (and improved!) by a single yarprobotinterface .xml running various fakeXXX devices.

drdanz, 2020-07-28:
@randaz81 should we get rid of that instead?
I think it is used in some examples though:

https://github.com/robotology/yarp/blob/2b123c2503ce48710754717e9dae2c6b4621b126/example/dev_check/main.cpp#L24-L30

There is also an outdated tutorial that creates another `fakebot`

https://github.com/robotology/yarp/blob/2b123c2503ce48710754717e9dae2c6b4621b126/example/tutorial/fakebot/FakeBot.h#L24

drdanz, 2020-07-28:
both `test_grabber` and `test_motor` were renamed without deprecation in YARP 3.4.0.

closed 2020-12-22 by drdanz