robotology/yarp #2352: Move ovrheadset YARP device to a separate standalone repository.

traversaro, 2020-09-04:
To simplify installation of the software on Windows with the Oculus SDK, it would be great to move the ovrheadset YARP device to a separate repository. It could be named `yarp-oculus-device` or `yarp-vr-devices` if in the future we want to also add OpenXR-based devices.

drdanz, 2020-09-25:
I imported the ovrheadset here:
https://github.com/robotology/yarp-device-ovrheadset

There are probably still a couple of things to fix, but the code is officially there, and the `src/devices/ovrheadset` in YARP is therefore frozen.

I will remove it from YARP as soon as I am sure that it works properly.

CC @S-Dafarra @GiulioRomualdi

drdanz, 2020-12-14:
@S-Dafarra Any news on this? Did you manage to test it?

traversaro, 2020-12-14:
The superbuild was switched to use the new repo in https://github.com/robotology/robotology-superbuild/pull/518 .

S-Dafarra, 2020-12-14:
> @S-Dafarra Any news on this? Did you manage to test it?

I just compiled it, but I did not test it in practice. What would be the steps to move from the old YARP device to the new external one?

traversaro, 2020-12-14:
> > @S-Dafarra Any news on this? Did you manage to test it?
> 
> I just compiled it, but I did not test it in practice. What would be the steps to move from the old YARP device to the new external one?

They are drop in compatible, so I guess that if you disable the YARP one, do a clean install, and then install the new one, no further modification should be necessary.

commit cba38a54 (2021-01-21): Remove ovrheadset device driver

Moved to https://github.com/robotology/yarp-device-ovrheadset

Fixes #2352
commit 0e9793be (2021-01-28): Remove ovrheadset device driver

Moved to https://github.com/robotology/yarp-device-ovrheadset

Fixes #2352
commit a4da959a (2021-01-28): Remove ovrheadset device driver

Moved to https://github.com/robotology/yarp-device-ovrheadset

Fixes #2352
PR #2465 (2021-01-28): Remove ovrheadset device driver
Moved to https://github.com/robotology/yarp-device-ovrheadset

Fixes #2352
closed 2021-01-28 by drdanz