robotology/yarp #2355: Yarpmanager buttons disabled when changing application tab

marco-monforte, 2020-09-10:
**Describe the bug**
Once you change the application tab in the yarpmanager, the buttons Play/Stop/Kill/Connect/Disconnect (both single module and for all modules) are disabled forever.

**To Reproduce**
Steps to reproduce the behavior:
1) Load an application from Entities -> The buttons are ok
2) Load a second application -> The buttons are still ok
3) Move back to the first application -> Buttons are disabled

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

- When loading the first application:

[image]

- After loading a second application and moving back to the first tab:

[image]

**Configuration (please complete the following information):**
 - OS: Ubuntu 18.04.5 LTS
 - yarp version: YARP version 3.4.100+20200910.2+git1f9456379
-  compiler: gcc version 7.5.0

pattacini, 2020-09-19:
I can confirm I've seen the same problem.

Nicogene, 2020-09-21:
This issue is under investigation and the function causing this bug is the following:

https://github.com/robotology/yarp/blob/8ea25d2647b649b290be83a54ce0f2d094b59269/src/yarpmanager/src-manager/mainwindow.cpp#L909

In particular, there is this piece of code https://github.com/robotology/yarp/blob/8ea25d2647b649b290be83a54ce0f2d094b59269/src/yarpmanager/src-manager/mainwindow.cpp#L958-L969

Removing this should fix the issue but first I have to understand what was the rationale behind it before trimming it.

Nicogene, 2020-09-21:
The `*_all` actions seem  to be handled only inside https://github.com/robotology/yarp/blob/8ea25d2647b649b290be83a54ce0f2d094b59269/src/yarpmanager/src-manager/mainwindow.cpp (as expected)

And here is the history of that file https://github.com/robotology/yarp/commits/8ea25d2647b649b290be83a54ce0f2d094b59269/src/yarpmanager/src-manager/mainwindow.cpp

The only recent commit is https://github.com/robotology/yarp/commit/41473510bc42960def8ffab8a8cdee6f01a558ad#diff-08d198d7b174504e27ad5e7f2cb2d19c that to me doesn't seem to interact with the buttons in the sidebar

Nicogene, 2020-09-21:
It seems to me that there is another hidden issue, the refresh button is enabled when opening a new app(see https://github.com/robotology/yarp/blob/8ea25d2647b649b290be83a54ce0f2d094b59269/src/yarpmanager/src-manager/mainwindow.cpp#L644) But then is disabled when changing tab if the tab opened **IS NOT** a `Resource`(then if you switch between applications it get disabled)

https://github.com/robotology/yarp/blob/8ea25d2647b649b290be83a54ce0f2d094b59269/src/yarpmanager/src-manager/mainwindow.cpp#L951-L957

I have no idea what is the rationale behind this

pattacini, 2020-09-21:
@Nicogene I would verify that this behavior is actually as you've read from the code.
Then, I would amend the code to make it behave as a traditional UX should prescribe.

Nicogene, 2020-09-21:
> @Nicogene I would verify that this behavior is actually as you've read from the code.
Then, I would amend the code to make it behave as a traditional UX should prescribe.

Yes I first noticed it testing the manager

I checked out to `v3.2.0` and those bugs are not present, I will try with `git bisect`

Nicogene, 2020-09-21:
Bisecting I found out that here is the faulty commit:

```
41473510bc42960def8ffab8a8cdee6f01a558ad is the first bad commit
commit 41473510bc42960def8ffab8a8cdee6f01a558ad
Author: Nicolo <nicolo.genesio@iit.it>
Date:   Fri Feb 28 15:30:43 2020 +0100

    yarpmanager: add builder tab.

 .../src-manager/applicationviewwidget.cpp          |  43 -------
 .../src-manager/applicationviewwidget.h            |   5 -
 .../src-manager/applicationviewwidget.ui           | 139 +++++++++++----------
 src/yarpmanager/src-manager/mainwindow.cpp         |  97 --------------
 src/yarpmanager/src-manager/mainwindow.h           |   2 -
 5 files changed, 74 insertions(+), 212 deletions(-)
```

In particular I trimmed away an `else` statement. Now the PR contains the right fix

drdanz, 2020-09-22:
Fixed by @Nicogene in #2367

PR #2367 (2020-09-22): Fix yarpmanager tabs/buttons issue
This PR fixes the issue that disables side menubar buttons when changing tab.

It fixes #2355

Please review code.
commit 39834830 (2020-09-22): yarmanager: fix issue that disable buttons when changing tab

It fixes #2355
closed 2020-09-22 by drdanz