robotology/yarp #2365: os/Stamp should be using a fixed size type for counter

drdanz, 2020-09-16:
The `yarp::os::Stamp` class uses `int` to represent the counter.
Since `int` has a different representation depending on the system, it should be using a fixed size type, for example `int32_t`.

Also it should probably use an unsigned type, i.e. `uint32_t`

drdanz, 2020-09-16:
cc @Nicogene @traversaro

pattacini, 2020-09-16:
@drdanz there are conditions where a negative counter (specifically `-1`) is employed to mean "invalid" or "not-available".

Nicogene, 2020-09-16:
> @drdanz there are conditions where a negative counter (specifically -1) is employed to mean "invalid" or "not-available".

Exactly, if the Stamp is not defined it has -1 has seq number and 0.0 has ts, changing it would be a break of ABI

drdanz, 2020-09-16:
Technically, if only `-1` is used, it could be possible to use `uint32_t` and define

```
getMaxCount() == std::numeric_limits<uint32_t>::max() - 1;
```

since `-1` has the exact representation as `std::numeric_limits<uint32_t>`

(btw `getMaxCount` should be `constexpr`)

pattacini, 2020-09-16:
>since -1 has the exact representation as std::numeric_limits<uint32_t>

Yeah, but then how this relates to checks like this below:

https://github.com/robotology/yarp/blob/71097ed9b24369d2709dc37298120130cc5c4868/src/libYARP_os/src/yarp/os/Stamp.cpp#L41-L44

If `sequenceNumber` is unsigned, then we need to change the check.
I'm afraid that similar checks might be spread around.

drdanz, 2021-07-02:
This was fixed in `yarp::os::Header` which will gradually replace `yarp::os::Stamp`. There is no longer need to change it in Stamp as well.

closed 2021-07-02 by drdanz