robotology/yarp #2371: Move realsense2 YARP devices to a separate standalone repository.

drdanz, 2020-09-25:
The realsense2 device depends on the `realsense2` library that is not available by default in debian/ubuntu.
In order to simplify the installation using binary packages, we should move this device in a separate repository.

drdanz, 2020-09-25:
I imported the realsense2 device here:
https://github.com/robotology/yarp-device-realsense2

There are probably still a couple of things to fix, but the code is officially there, and the `src/devices/realsense2` in YARP is therefore frozen.
Also the documentation in [`doc/subdevice_realsense.md`](https://github.com/robotology/yarp/blob/303a8f5dea8ea214e343772f8b97f9dacb7b082f/doc/subdevice_realsense.md), should probably be moved in the new repository (cc @aikolina)

I will remove it from YARP as soon as I am sure that it works properly.

xEnVrE, 2020-10-07:
> I will remove it from YARP as soon as I am sure that it works properly.

Fyi, I tested the new repository and everything seems ok. I tested with a D415 only.

[image]

drdanz, 2020-10-07:
Thanks @xEnVrE for testing!

PeterBowman, 2020-10-08:
Also successfully tested with a D435i (*realsense2* device), YARP 3.4.1, RealSense SDK 2.39.0.

drdanz, 2020-10-12:
Thanks @PeterBowman 
I think we are ready to remove the driver from master then...

drdanz, 2020-10-12:
>  Also the documentation in [`doc/subdevice_realsense.md`](https://github.com/robotology/yarp/blob/303a8f5dea8ea214e343772f8b97f9dacb7b082f/doc/subdevice_realsense.md), should probably be moved in the new repository

We should still figure out what to do with this documentation page, since if we move it out of YARP it will no longer be generated, but it doesn't make sense to leave it there, since the device is no longer there.
Perhaps we can convert it to markdown, and adapt it to be the `README.md` file in the new repository?

traversaro, 2020-10-12:
For small repos, a well written README is definitely the most accessible and maintainable form of documentation.

drdanz, 2020-10-12:
Done, updated the PR

PR #2383 (2020-10-13): Remove realsense2 device driver
Moved to https://github.com/robotology/yarp-device-realsense2

Fixes #2371
commit 92337ccd (2020-10-13): Remove realsense2 device driver

Moved to https://github.com/robotology/yarp-device-realsense2

Fixes #2371
closed 2020-10-13 by drdanz