robotology/yarp #2372: Add a documentation page for (known) devices in other repositories

drdanz, 2020-09-25:
Since we started moving a few devices to external repositories (#2352 and #2371), we should, at least, have a documentation page with index with the list of known devices developed in external repositories.

I'll start the list, please add the other devices in this issue, and I'll add them to the list

- [ovrheadset](https://github.com/robotology/yarp-device-ovrheadset)
- [realsense2](https://github.com/robotology/yarp-device-realsense2)
- [xsensmt](https://github.com/robotology/xsensmt-yarp-driver)

traversaro, 2020-10-07:
I think there is no problem in renaming https://github.com/robotology/xsensmt-yarp-driver in yarp-device-xsensmt for consistency.

jgvictores, 2020-10-15:
Hi! I understand the origin of this issue resides in the reorganization due to the issues cited above, so sorry for the intrusion. ^^

Just for the sake of completeness, I'd point to the https://github.com/roboticslab-uc3m/yarp-devices repository. While there is some software that is relatively specific to our setup, some of the supported physical devices are quite popular, for instance:
- https://github.com/roboticslab-uc3m/yarp-devices/tree/master/libraries/YarpPlugins/AravisGigE
- https://github.com/roboticslab-uc3m/yarp-devices/tree/master/libraries/YarpPlugins/Jr3 (pending https://github.com/roboticslab-uc3m/yarp-devices/issues/240)

I think we'd be happy to reorganize/decouple to further comply with your own best practices.

traversaro, 2020-10-15:
Hi @jgvictores, actually in the meeting were we discussed about this I pointed out that your repo was a good example of another repo that should have included in the list, so thanks for sharing!

drdanz, 2020-11-18:
xsensmt was renamed: see robotology/yarp-device-xsensmt#24