robotology/yarp #2378: ACE used on appveyor old (6.3.4)

jwillemsen, 2020-10-07:
**Describe the bug**

The ACE version used on the appveyor CI is old, it is 6.3.4 which is from 2017, see https://github.com/DOCGroup/ACE_TAO/releases/tag/ACE%2BTAO-6_4_3. It was formally never tested for Visual Studio 2019.

Would recommend to upgrade ACE on appveyor, maybe use the conda or vcpkg packaged version as alternative?

See https://github.com/robotology/yarp/blob/f8773835064a25848010a934d9366711a0fc305c/.appveyor.yml#L58

drdanz, 2020-10-07:
Fixed in fbc3bbd1e2f969b4bd2f45a21e23605961d5f3aa (now using ACE 6.5.6)

closed 2020-10-07 by drdanz

jwillemsen, 2020-10-08:
FYI, with github actions there is build in support for vcpkg caching, we are using that for ACE, see https://github.com/DOCGroup/ACE_TAO/blob/643377c9a6a544c7fde64a56aff3c31c55bb8813/.github/workflows/windows.yml#L99.

commit fbc3bbd1 (2020-10-07): ci/appveyor: Download and use dependencies from robotology-superbuild-dependencies

Fixes #2378