robotology/yarp #2388: yarphear generates a flickering audio if it runs from yarpmanager

GiulioRomualdi, 2020-10-20:
**Describe the bug**
When I run `yarphear` from `yarpmanager` on Ubuntu 18.04 the audio flickers. Checking the logger  the following messages are displayed
```
1351876 51,203914 <UNDEFINED> ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
 
1351877 51,206950 <UNDEFINED> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
 
1351878 51,207024 <UNDEFINED> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
 
1351879 51,207052 <UNDEFINED> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
 
1351880 51,207074 <UNDEFINED> ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
 
1351881 51,224133 <UNDEFINED> ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
 
 51,244309 <INFO> Created device <portaudioPlayer>. See C++ class PortAudioPlayerDeviceDriver for documentation. 
 51,305018 <INFO> Port /icub/yarphear:i active at tcp://10.0.0.164:10027/ 
 51,305109 <INFO> Type "help" for usage 
 51,305170 <INFO> Muted 
 51,305200 <INFO> Type "help" for usage 
 51,305225 <INFO> Audible again 
 51,305248 <INFO> Type "help" for usage 
 51,305270 <INFO> Muted 
 51,305291 <INFO> Type "help" for usage 
 51,305314 <INFO> Audible again 
 51,305335 <INFO> Type "help" for usage 
```

If I run the very same command from the terminal the application works as expected. 
I think that the error is generated by these lines 
https://github.com/robotology/yarp/blob/af2a17deda80d5cc317cf6f929c4967a1d6b3197/src/yarphear/yarphear.cpp#L229-L231

**To Reproduce**
Run the following application from the manager
```xml
<!-- Copyright (C) 2020 Fondazione Istituto Italiano di Tecnologia (IIT) -->
<!-- All Rights Reserved. -->
<!-- Author: Giulio Romualdi <giulio.romualdi@iit.it> -->

<application>

  <name>Audio for Retargeting</name>
  <description>Audio application.</description>
  <version>1.0</version>

  <!-- Devices -->
  <module>
    <name>yarpdev</name>
    <parameters> --device AudioRecorderWrapper --subdevice portaudioRecorder --name /icub/microphone  --min_samples_over_network 4000  --max-samples_over_network 4000  --rate 160000 --samples 16000</parameters>
    <node>icub-console-gui</node>
  </module>

  <module>
    <name>yarpdev</name>
    <parameters> --device AudioRecorderWrapper --subdevice portaudioRecorder --name /virtualizer/microphone  --min_samples_over_network 4000  --max-samples_over_network 4000  --rate 160000 --samples 16000</parameters>
    <node>icub-virtualizer</node>
  </module>

  <module>
    <name>yarphear</name>
    <parameters>--name /icub/yarphear:i</parameters>
    <node>icub-console-gui</node>
  </module>

   <module>
    <name>yarphear</name>
    <parameters>--name /virtualizer/yarphear:i</parameters>
    <node>icub-virtualizer</node>
  </module>

  <!-- Connections -->
  <connection>
    <from>/virtualizer/microphone/audio:o</from>
    <to>/icub/yarphear:i</to>
  </connection>

  <connection>
    <from>/icub/microphone/audio:o</from>
    <to>/virtualizer/yarphear:i</to>
  </connection>

</application>
```

Here you can also find the complete log 

[log_icub-console-gui_yarphear_18321.txt](https://github.com/robotology/yarp/files/5410019/log_icub-console-gui_yarphear_18321.txt)

traversaro, 2020-10-20:
What is the output if you run it from the terminal?

GiulioRomualdi, 2020-10-20:
> What is the output if you run it from the terminal?

``` 
yarphear --name /icub/yarphear:i 
ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_route.c:867:(find_matching_chmap) Found no matching channel map
[INFO]created device <portaudioPlayer>. See C++ class PortAudioPlayerDeviceDriver for documentation.
yarp: Port /icub/yarphear:i active at tcp://10.0.0.164:10027/ 
[INFO]Type "help" for usage
>>
```

GiulioRomualdi, 2020-10-20:
I'm obtaining the same behavior if I press `enter` on the terminal running `yarphear`
```
[INFO]Type "help" for usage
>>↵
[INFO]Muted
[INFO]Type "help" for usage
>>↵
[INFO]Audible again
[INFO]Type "help" for usage
>>↵
[INFO]Muted
[INFO]Type "help" for usage
>>
[INFO]Audible again
[INFO]Type "help" for usage
>>↵
[INFO]Muted
[INFO]Type "help" for usage
>>↵
[INFO]Audible again
[INFO]Type "help" for usage
>>
```
Here I used `↵` to indicate the `enter`

lornat75, 2020-10-20:
Can you try:
- redirecting stdio, e.g. adding <stdio>node3</stdio> in the xml file
- executing yarp run manually on the console of icub-console-gui, e.g. yarp run --server /icub-console-gui

traversaro, 2020-10-20:
> executing yarp run manually on the console of icub-console-gui, e.g. yarp run --server /icub-console-gui

In this direction you should also try to launch yarphear via the yarpmanager but not using the `yarprun` deployer, i.e. running it on the special `localhost` node.

randaz81, 2020-10-21:
Just a side comment: 
Since you already have in your .xml `yarpdev --device AudioRecorderWrapper --subdevice portaudioRecorder`, the most obvious (and modern) choice to transfer audio is to use `yarpdev   --device AudioPlayerWrapper --subdevice portaudioPlayer`. You probably do not need yarphear at all (unless you want to save audio to disk)

S-Dafarra, 2020-10-21:
> Since you already have in your .xml `yarpdev --device AudioRecorderWrapper --subdevice portaudioRecorder`, the most obvious (and modern) choice to transfer audio is to use `yarpdev --device [...]

I did not understand. What would be the module responsible to stream the audio to the speakers? What would be the resulting ``yarpmanager`` application you had in mind?

GiulioRomualdi, 2020-10-21:
Thank you @randaz81 your solution (https://github.com/robotology/yarp/issues/2388#issuecomment-713418289) works for us.
Indeed I've created a `yarpmanager` application in: https://github.com/robotology/walking-teleoperation/pull/41

closed 2021-02-17 by drdanz