robotology/yarp #2389: Alternative to the --verbose option

S-Dafarra, 2020-10-22:
In ``yarp-3.4`` the ``--verbose`` option is deprecated. I was finding this feature particularly handy when using the ``ResourceFinder`` as it was helping to debug where a certain file was searched. Nevertheless, if I run ``yarp resource --find something.ini --verbose``
```
yarp resource --find something.ini --verbose
[WARNING] |yarp.os.ResourceFinder| The 'verbose' option is deprecated.
[WARNING] |yarp.os.ResourceFinder| The 'verbose' option is deprecated.
[ERROR] |yarp.os.Property| cannot read from something.ini
""
```
I simply get a warning, but no alternative and no verbose output. Is there a way to circumvent this?

PeterBowman, 2020-10-22:
The `verbose` option was superseded by logging components in https://github.com/robotology/yarp/commit/301926c2f67c26dba4a7fd28930de99d7ad8c4cc, here and in several other places around YARP's codebase (search for deprecations in the [v3.4.0 release notes](https://github.com/robotology/yarp/releases/tag/v3.4.0)).

I see the helper [`toDox` method](https://github.com/robotology/yarp/blob/45f16299970647a24d0f711d6983b3da4363f730/src/libYARP_dev/src/yarp/dev/Drivers.cpp#L312-L356) is still there, this is how the verbose output was being printed to console. Now, it takes advantage of `yCDebug` macros. According to [*Logging with YARP docs*](http://www.yarp.it/yarp_logging.html#autotoc_md645), you should set the `YARP_VERBOSE` environment variable to `1` in order to enable debug lines in YARP itself.

S-Dafarra, 2020-10-23:
Thanks a lot @PeterBowman, that was exactly what I was looking for.

closed 2020-10-23 by S-Dafarra