robotology/yarp #2396: Access to timestamp not protected with mutex

drdanz, 2020-11-02:
I think that this write access to the `time` variable should be protected using the `timeMutex` which, from my understanding, is there just to protect this variable.

https://github.com/robotology/yarp/blob/a64b8f96ab1957c3fa7ec298eebc718886877834/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L894

This might cause some invalid timestamp or similar random issues.

Can anyone who knows the code check if I'm right? Perhaps @randaz81 or @traversaro?

drdanz, 2020-11-03:
https://github.com/robotology/yarp/blob/a64b8f96ab1957c3fa7ec298eebc718886877834/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L4330-L4335

drdanz, 2020-11-03:
https://github.com/robotology/yarp/blob/a64b8f96ab1957c3fa7ec298eebc718886877834/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L935

drdanz, 2020-11-03:
https://github.com/robotology/yarp/blob/a64b8f96ab1957c3fa7ec298eebc718886877834/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L943

https://github.com/robotology/yarp/blob/a64b8f96ab1957c3fa7ec298eebc718886877834/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L964

traversaro, 2020-11-03:
I agree, the unprotected access of time is a bug.

PR #2397 (2020-11-03): devices/controlboardwrapper2: Fix access to time not protected by mutex
# Devices

#### `controlboardwrapper2`

Fix access to the timestamp not protected by mutex (#2396).

---------------------------

Instead of repeatedly accessing the `time` variable, I saved it in while protected by the mutex, and used the same timestamp for the whole `run` function.

Fixes #2396
commit 924476d7 (2020-11-03): devices/controlboardwrapper2: Fix access to time not protected by mutex

Fixes #2396
closed 2020-11-03 by drdanz