robotology/yarp #2400: yarpdataplayer --hidden cannot start on headless system

arrenglover, 2020-11-06:
**Describe the bug**
when running `yarpdataplayer --hidden` on a system without X11 server the following error occurs:

```
qt.qpa.xcb: could not connect to display 172.31.32.1:0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Aborted
```

**To Reproduce**
- make sure no X11 server is running (xming, vcxsvr etc.)
- run bash for ubuntu
- `yarpserver --write`
- in another terminal `yarpdataplayer --hidden` 

**Expected behavior**
I'm not sure if the `--hidden` flag is meant to make it run on a headless system but I would hope the yarpdataplayer can be used without X11. Just using the RPC to load, start, and stop the player.

**Configuration (please complete the following information):**
 - OS: Windows 10 19042  + ubuntu 20.04 on WSL2
 - yarp version: 3.4.1
-  compiler:

--   CMake: 3.16.3 (/usr/bin/cmake)
--   C Compiler: GNU 9.3.0 (/usr/bin/cc)
--   CXX Compiler: GNU 9.3.0 (/usr/bin/c++)

**Additional context**
 I'm testing on WSL, but hoped to use it on a server without any graphics capabilities.

vtikha, 2020-11-06:
Thanks for reporting @arrenglover, I will look at it and will try to come up with a solution.

arrenglover, 2020-11-06:
thanks @vtikha. I'm actually trying to get this to work for an external group so they can do some testing *this weekend* 😱. I will look into it myself, if you can give me any pointers on where to look in the yarpdataplayer - or a quick hack to get it working immediately I would appreciate it. I think I would also need #2401 solved 🤔

vtikha, 2020-11-06:
@arrenglover, ok, then we can try to look at it in parallel. Indeed it might uncover some mysteries for for #2401. 
Thanks @arrenglover

traversaro, 2020-11-06:
If you need a quick fix, getting a X server to run on Windows and make sure that WSL2 is configured to be able to connect to it may work, the necessary procedure is documented in https://github.com/robotology/robotology-superbuild#run-graphical-applications-on-wsl2 .

arrenglover, 2020-11-06:
thanks @traversaro - yes I have an X server and can run yarpdataplayer without problems. I'm using this as an example for a system external (for which I don't have any details) for which they can't get an X server to work.

traversaro, 2020-11-06:
Ack, I see!

arrenglover, 2020-11-09:
@vtikha - FYI I think the other group got something working with an X server. So we aren't being blocked by this issue, no need to rush.

vtikha, 2020-11-09:
Good, as I was not able to have a look at it over the weekend. I believe that in any case this has to be fixed asap to work correctly without an X server. In this respect, in the next couple of days I will modify things so that we have a `yarpdataplayer-console` version alongside the `yarpdataplayer` as @apaikan implemented for the `yarpmanager`. This way, we can remove the `--hidden` option and be consistent with the other main `GUI` of `YARP`.

drdanz, 2020-11-09:
I had a similar problem long time ago, `QApplication` requires an X server (or equivalent) running.
The [`QApplication`](https://doc.qt.io/qt-5/qapplication.html#details) documentation explains how to handle an executable that can run with or without GUI.

Anyway I think that here the issue is that all the logic is inside the main window, and I'm not sure that you can create one (even if hidden) with a `QCoreApplication`, a larger refactor might be necessary.

vtikha, 2020-12-22:
Full refactoring has been done as discussed above. The PR can be found here done: [here](https://github.com/robotology/yarp/pull/2442)
cc @arrenglover

closed 2021-01-12 by drdanz