robotology/yarp #2411: Support yarp::sig::VectorOf<int> in Python

xEnVrE, 2020-11-17:
**Is your feature request related to a problem? Please describe.**

I would like to use `yarp::sig::VectorOf<int>` in Python. It seems that [yarp.i](https://github.com/robotology/yarp/blob/master/bindings/yarp.i) is missing that specialization of `yarp::sig::VectorOf` for integers.

**Additional context**

I tried to replicate all the recipes already implemented in `yarp.i` to support the `yarp::sig::VectorOf<double>` and I was able to make it work (I only tested in Python though). The changes are in [this commit](https://github.com/xEnVrE/yarp/commit/fd3a09b10a1088fada06b4accd4db8382729540a). However, I am not sure whether all these changes are required and sound.

Is it ok if I open a PR?

xEnVrE, 2020-11-17:
cc @drdanz @diegoferigo

traversaro, 2020-11-17:
> However, I am not sure whether all these changes are required and sound.

Assessing that is exactly the point of opening a PR, so I would say go forward and let's see the changes in a PR, thanks (even if probably given the state of the CI @drdanz could have other ideas).

drdanz, 2020-11-17:
The CI using github actions is *almost* working, feel free to open the PR, I will eventually test it locally, or delay it for a couple of days in order to test it properly... Thanks!

PR #2412 (2020-11-20): Add support for yarp::sig::VectorOf<int> in bindings/yarp.i
As per title this PR add supports for `yarp::sig::VectorOf<int>` in `bindings/yarp.i`.

Fixes #2411
closed 2020-11-20 by drdanz