robotology/yarp #2414: usbCameraRaw YARP device is not installed correctly in Linux

traversaro, 2020-11-19:
**Describe the bug**
The `usbCameraRaw` plugin installed by the YARP option `ENABLE_yarpmod_usbCamera` does not work correctly, and it is not shown correctly by the `yarpdev --list` .

**To Reproduce**
Install YARP and enable the usbCamera device. Then, run `yarpdev --list`, you get:
~~~
 [INFO] |yarp.dev.Drivers| Here are devices listed for your system:
Error:  |yarp.os.YarpPluginSettings| Error while opening /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/yarp/yarp_usbCamera.so:
  /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/yarp/yarp_usbCamera.so: undefined symbol: usbCameraRaw
Error:  |yarp.os.YarpPluginSettings| Error while opening /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/yarp/yarp_usbCamera.so:
  /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/yarp/yarp_usbCamera.so: undefined symbol: usbCameraRaw
Error:  |yarp.os.YarpPluginSettings| Error while opening /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/yarp/yarp_usbCamera.so:
  /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/yarp/yarp_usbCamera.so: undefined symbol: usbCameraRaw
Error:  |yarp.os.YarpPluginSettings| Error while opening /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/yarp/yarp_usbCamera.so:
  /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/lib/yarp/yarp_usbCamera.so: undefined symbol: usbCameraRaw
Warning:  |yarp.dev.Drivers| Wrong library name for plugin usbCameraRaw
~~~

**Expected behavior**
I would expect `yarpdev --list`  to work correctly as it does for all the other YARP devices. 

**Screenshots**

[image]

**Configuration (please complete the following information):**
 - OS: Ubuntu 20.04
 - yarp version: both 3.4.0, 3.4 branch and master
-  compiler: GCC 9.3 

**Additional context**
I discovered this in https://github.com/robotology/robotology-superbuild/pull/521/checks?check_run_id=1425069998 .

drdanz, 2020-11-19:
Did you enable also `ENABLE_yarpmod_usbCameraRaw`? I don't know the difference, but they are two different plugins in the same library

traversaro, 2020-11-19:
I added it in https://github.com/robotology/robotology-superbuild/pull/521/commits/7b9367a5a05f9118048a9fde0b2e56ad72c6df8f, let's see if it fixes the issue.

traversaro, 2020-11-20:
Indeed, enabling `ENABLE_yarpmod_usbCameraRaw` everything works as expected. I guess it is a bit counter intuitive that if you enable  `ENABLE_yarpmod_usbCamera`  and not `ENABLE_yarpmod_usbCameraRaw` you get a non-working `usbCameraRaw` installed, but for sure that is another problem.

closed 2020-11-20 by traversaro