robotology/yarp #2420: Unable to include LogStream with CasAdi

Giulero, 2020-11-25:
**Describe the bug**

If I include 
```cpp
#include <casadi/casadi.hpp>
#include <yarp/os/LogStream.h>
```
the following error occurs while compiles:

```
In file included from /home/giuseppe/robotology-superbuild/build/install/include/yarp/os/all.h:26,
                 from /home/giuseppe/Desktop/IronCub/element_transition-maneouvers/modules/point_mass_opt_test/pointMassOptimizer/include/Solver.h:13,
                 from /home/giuseppe/Desktop/IronCub/element_transition-maneouvers/modules/point_mass_opt_test/pointMassOptimizer/src/Solver.cpp:9:
/home/giuseppe/robotology-superbuild/build/install/include/yarp/os/LogStream.h:238:22: error: redefinition of ‘template<class T> std::ostream& std::operator<<(std::ostream&, const std::vector<T>&)’
  238 | inline std::ostream& std::operator<<(std::ostream& os, const std::vector<T>& t)
      |                      ^~~
In file included from /home/giuseppe/git_repos/casadi/build/install/include/casadi/core/matrix_fwd.hpp:29,
                 from /home/giuseppe/git_repos/casadi/build/install/include/casadi/core/sx_elem.hpp:33,
                 from /home/giuseppe/git_repos/casadi/build/install/include/casadi/core/core.hpp:30,
                 from /home/giuseppe/git_repos/casadi/build/install/include/casadi/casadi.hpp:29,
                 from /home/giuseppe/Desktop/IronCub/element_transition-maneouvers/modules/point_mass_opt_test/pointMassOptimizer/include/Solver.h:12,
                 from /home/giuseppe/Desktop/IronCub/element_transition-maneouvers/modules/point_mass_opt_test/pointMassOptimizer/src/Solver.cpp:9:
/home/giuseppe/git_repos/casadi/build/install/include/casadi/core/casadi_misc.hpp:430:12: note: ‘template<class T> std::ostream& std::operator<<(std::ostream&, const std::vector<T>&)’ previously declared here
  430 |   ostream& operator<<(ostream& stream, const vector<T>& v) {
      |            ^~~~~~~~
make[2]: *** [modules/point_mass_opt_test/pointMassOptimizer/CMakeFiles/TrajOptSolver.dir/build.make:83: modules/point_mass_opt_test/pointMassOptimizer/CMakeFiles/TrajOptSolver.dir/src/Solver.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:361: modules/point_mass_opt_test/pointMassOptimizer/CMakeFiles/TrajOptSolver.dir/all] Error 2
make: *** [Makefile:150: all] Error 2
```

**Configuration (please complete the following information):**
 - OS: Ubuntu 20.04
 - yarp version: 3.4
-  compiler: gcc version 9.3.0

GiulioRomualdi, 2020-11-25:
Unfortunately, both `yarp` and `casadi` expands `std` namespace

### casadi
https://github.com/casadi/casadi/blob/fadc86444f3c7ab824dc3f2d91d4c0cfe7f9dad5/casadi/core/casadi_misc.hpp#L428-L433

### yarp
https://github.com/robotology/yarp/blob/13d2bc8ea92621b29055d10c6dbc43809f35c1b0/src/libYARP_os/src/yarp/os/LogStream.h#L28-L31

Probably the fastest solution is avoid to use LogStream in your application

traversaro, 2020-11-25:
Thanks for opening the issue! We had a similar discussion in https://github.com/robotology/yarp/issues/2067 .

drdanz, 2020-11-25:
That's the same bug, I'm closing this in favour of #2067

closed 2020-11-25 by drdanz