robotology/yarp #2435: Math formulas are not correctly displayed in Doxygen documentation

traversaro, 2020-12-17:
**Describe the bug**
Math documentation in http://www.yarp.it/git-master/classyarp_1_1dev_1_1IOrientationSensors.html#adb2a40d03c747aa9c8b7fc542f193fd9 is not correctly render, probably due to some missing option passed to doxygen during the generation process.

**To Reproduce**
Visit http://www.yarp.it/git-master/classyarp_1_1dev_1_1IOrientationSensors.html#adb2a40d03c747aa9c8b7fc542f193fd9 and see raw latex formulas .

**Expected behavior**
The math formula rendered, see http://mathb.in/48272 .

**Configuration (please complete the following information):**
 - OS: Windows
 - yarp version: Not Relevant
-  compiler: Not Relevant

traversaro, 2020-12-17:
cc @Nicogene

PeterBowman, 2020-12-17:
One way to handle this is through Doxygen's [`CITE_BIB_FILES`](https://www.doxygen.nl/manual/config.html#cfg_cite_bib_files) property: [inline comment](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/41a28707fc2ffb26122068e55eed6550929cdced/libraries/ScrewTheoryLib/ScrewTheoryIkProblem.hpp#L41-L76), [HTML+LaTeX render](http://robots.uc3m.es/dox-kinematics-dynamics/classroboticslab_1_1ScrewTheoryIkSubproblem.html#ad80bdbafabb6a6a5ab69f0aec3ad9741). I'd suggest to take advantage of the modern [`FindDoxygen` module](https://cmake.org/cmake/help/latest/module/FindDoxygen.html) which takes care of generating a `Doxyfile` from CMake configuration and creating a new target that calls `doxygen` to parse it: [example](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/41a28707fc2ffb26122068e55eed6550929cdced/doc/CMakeLists.txt). That is, no more [`Doxyfile.in`](https://github.com/robotology/yarp/blob/0481f994c6e03897d038c5f1d1078145646a1772/doc/Doxyfile.in).

traversaro, 2020-12-17:
Thanks for chiming in @PeterBowman !

drdanz, 2020-12-21:
I believe this could be due to some missing latex package in the documentation build environment, I'll move this to the robotology/yarp-documentation repository...