robotology/yarp #2457: Not able to communicate with two yarp ports opened in the same laptop

GiulioRomualdi, 2021-01-13:
**Describe the bug**
As written in the title I'm not able to communicate with two yarp ports opened in the same laptop. 

I also noticed that when I ran `yarpserver` an error is thrown 
```
yarpserver

     ==========================
    | __  __ ___  ____   ____  |
    | \ \/ //   ||  _ \ |  _ \ |
    |  \  // /| || |/ / | |/ / |
    |  / // ___ ||  _ \ |  _/  |
    | /_//_/  |_||_| \_\|_|    |
    |                          |
     ==========================

Call with --help for information on available options
Using port database: :memory:
Using subscription database: :memory:
IP address: default
Port number: 10000
[INFO] |yarp.os.Port| Port /icub09 active at tcp://172.19.0.1:10000/
Registering name server with itself
[ERROR] |yarp.os.impl.DgramTwoWayStream| cannot connect to multi-cast address
 * register "/icub09" tcp "172.19.0.1" 10000
   + set "/icub09" offers http name_ser local tcp fast_tcp mcast udp text text_ack bayer human mjpeg portmonitor priority rossrv shmem tcpros unix_stream xmlrpc
   + set "/icub09" accepts http name_ser local tcp fast_tcp mcast udp text text_ack bayer human mjpeg portmonitor priority rossrv shmem tcpros unix_stream xmlrpc
   + set "/icub09" ips "127.0.0.1" "172.19.0.1" "172.18.0.1" "172.17.0.1" "::1" "fe80::42:a2ff:fe0a:548f%5" "fe80::982e:62ff:fe7a:4ce8%12"
   + set "/icub09" process 9733
 * register fallback mcast "224.2.1.1" 10000
   + set fallback offers http name_ser local tcp fast_tcp mcast udp text text_ack bayer human mjpeg portmonitor priority rossrv shmem tcpros unix_stream xmlrpc
   + set fallback accepts http name_ser local tcp fast_tcp mcast udp text text_ack bayer human mjpeg portmonitor priority rossrv shmem tcpros unix_stream xmlrpc
   + set fallback ips "127.0.0.1" "172.19.0.1" "172.18.0.1" "172.17.0.1" "::1" "fe80::42:a2ff:fe0a:548f%5" "fe80::982e:62ff:fe7a:4ce8%12"
   + set fallback process 9733
 * set "/icub09" nameserver "true"
Name server can be browsed at http://172.19.0.1:10000/
Ok. Ready!
```
`yarp check` fails with the following error  

```
yarp check
==================================================================
=== This is "yarp check"
=== It is a very simple sanity check for your installation
=== If it freezes, try deleting the file reported by "yarp conf"
=== Also, if you are mixing terminal types, e.g. bash/cmd.exe
=== on windows, make sure the "yarp conf" file is the same on each
==================================================================
=== Trying to register some ports
[INFO] |yarp.os.Port| Port /tmp/port/1 active at tcp://10.0.10.106:10002/
[INFO] |yarp.os.Port| Port /tmp/port/2 active at tcp://10.0.10.106:10003/
==================================================================
=== Trying to connect some ports
[ERROR] |yarp.os.Network| Cannot connect to port /tmp/port/2
==================================================================
=== Trying to write some data
==================================================================
=== Trying to read some data
=== Trying to read some data
=== Trying to read some data
==================================================================
=== Trying to close some ports

```

**To Reproduce**
1. run `yarpserver`
2. open a new terminal and type `yarp write /foo`
3. open another terminal and type `yarp read ... /foo`

**Configuration (please complete the following information):**
 - OS: Ubuntu 18.04
 - yarp version: `YARP version 3.4.1+81-20201214.6+git1b0060380`
-  compiler: gcc 7.5

traversaro, 2021-01-13:
Probably something interesting is happening with the different network interfaces on that laptop. Can you post the output of `ifconfig` ? Furthermore, I would try to set the `YARP_IP` env variable (http://www.yarp.it/git-master/yarp_env_variables.html) to force all YARP process to use a specific network interface, and see is the problem occurs in the same way

GiulioRomualdi, 2021-01-13:
Hi @traversaro we just noticed that `YARP_IP` was wrongly set (we are using this in the double network configuration)

closed 2021-01-13 by GiulioRomualdi