robotology/yarp #2480: Weird unreachable code in controlboardwrapper2

drdanz, 2021-02-03:
Given this check, and the return

https://github.com/robotology/yarp/blob/f7167f7bfbb9fd12b92dad12f334253a3165df43/src/devices/ControlBoardWrapper/SubDevice.cpp#L213-L217

the else in this part of code:

https://github.com/robotology/yarp/blob/f7167f7bfbb9fd12b92dad12f334253a3165df43/src/devices/ControlBoardWrapper/SubDevice.cpp#L231-L256

is unreachable. Also, if it was somehow reachable, the `pos->getAxes()` call at line 246 would crash the program, since `pos` is `nullptr`. I think that the original intention of the author could have been to call `getAxes` from some other interface, but here it makes no sense, since if `IPositionControl` is not available, the `open` function returns false. I think we can just remove this check, unless someone comes with a good explanation about why the code is there, and how it should be modified in order to do what it was supposed to do originally.

PeterBowman, 2021-02-03:
These are leftovers from the old `IPositionControl2` interface, see https://github.com/PeterBowman/yarp/commit/e6897835d1ce8e43ed2e88d34fbf7c27c1194d95 (in draft PR https://github.com/robotology/yarp/pull/2473) and linked commits in description.

drdanz, 2021-02-03:
Thanks @PeterBowman, that makes sense, I'll just leave those here for now, but I will remove them from the nws device...

randaz81, 2023-10-25:
controlboardwrapper2 has been deprecated

closed 2023-10-25 by randaz81