robotology/yarp #2492: portprefix and build attributes of robot tag of robotinterface XML

traversaro, 2021-02-11:
While working on, I noticed a few warnings: 
~~~
2: [WARNING] Invalid syntax while loading  at line 3 . "robot" element should contain the "build" attribute [unsigned int]. Assuming 0
2: [WARNING] Invalid syntax while loading  at line 3 . "robot" element should contain the "portprefix" attribute. Using "name" attribute
~~~

While I could easily silence them by attribute the required attributes, to be honest I am not sure what this attributes are/meant, who they are used and why they should be set. 

I tried to inspect the code, but here it is what I found:
* The `build` attribute seems to never been used (unless I am wrong). In this case, could we remove the warning if it is not set?
* The `portprefix` is used in https://github.com/robotology/yarp/blob/9b52a71e176e23ebe42956cf5b23d8ae40b255f0/src/yarprobotinterface/Module.cpp#L145 to overwrite the name attribute and it is used to specify the name of the YARP RPC port used to control the `yarprobotinterface`. However, at this point if this is the role of the `portprefix` argument, I don't know what is the role of the name attribute. If `portprefix` is not set, then `name` seems a perfectly fine fallback, so could it make sense to remove the warning also if the portprefix argument is missing?

traversaro, 2021-02-11:
cc @drdanz @randaz81

traversaro, 2021-02-11:
>  I don't know what is the role of the name attribute

Actually the get_name is exposed on the YARP rpc via https://github.com/robotology/yarp/blob/9b52a71e176e23ebe42956cf5b23d8ae40b255f0/src/yarprobotinterface/yarprobotinterface.thrift#L34 , so it may be useful to specify a portprefix different from the name.

drdanz, 2021-02-17:
* `build`: Long time ago, there was a plan to generate these files from some database. The build number was supposed to be increased by one in all files every time the files were re-generated, to avoid mixing files generated by 2 different generations.
Feel free to remove the warning if not set, but perhaps we should leave the check to ensure that, if set, the number is the same in every file.

* `portprefix`: I think that long time ago, we considered setting the names of the port using `YARP_PORT_PREFIX`, or something like that, considering that, instead of opening all devices with `--name /icub/right_arm`, we could have just passed `--name /right_arm` and the name would have been added automatically for all ports, reducing the number of errors, and increasing reusability of the files. Again, this was never implemented, so feel free to do whatever you want with it.