robotology/yarp #2496: yarpmotorgui failed to run

MrHuangWenjie, 2021-02-17:
Hi all,

I am new to yarp and iCub simulator. I have successfully installed yarp and ran yarpserver as well as basic functions like yarp read/ write. While, when I want to launch yarpmotorgui, the errors happened as:

**Error while opening device for part /icub/head.** and same for all parts.

the message stack ends with:

**announce "/yarpmotorgui/head/sequence:o"

unregister "/yarpmotorgui/head/sequence:o"**

Could you please give me advice on what may cause this?
thank you so much!

PeterBowman, 2021-02-17:
Hello. It looks like you are trying to connect to the real robot (`/icub` port prefix) while working on simulation. Try this:

```
yarpmotorgui --robot /icubSim
```

For completeness, this is the output of `yarpmotorgui --help` as of YARP 3.4.2:

```
[INFO] yarpmotorgui options:
[INFO] --robot <name>: name of the robot
[INFO] --parts ( <name1> <name2> ): parts of the robot to add to the list. (e.g. left_arm)
[INFO] --names ( <name1> <name2> ): full name of the ports of the robot to add to the list. (e.g. /icub/left_arm). This option is mutually exclusive with --robot --parts options
[INFO] --skip_parts ( <name1> <name2> ): parts of the robot to skip.
[INFO] --calib to enable calibration buttons (be careful!)
```

MrHuangWenjie, 2021-02-17:
> Hello. It looks like you are trying to connect to the real robot (`/icub` port prefix) while working on simulation. Try this:
>
> ```
> yarpmotorgui --robot /icubSim
> ```
>
> For completeness, this is the [...]

Hello, thank you for your reply!

However, unfortunately, the error still happened. Would that be problems during my installation even though I successfully ran yarpserver?

traversaro, 2021-02-17:
Is a simulator of iCub running? If not, I think that that outcome is expected. Furthermore, how did you installed YARP? On which operating system?

MrHuangWenjie, 2021-02-17:
> Is a simulator of iCub running? If not, I think that that outcome is expected. Furthermore, how did you installed YARP? On which operating system?

I think not as I just start the yarpserver before I wanted to run yarpmotorgui;

I installed yarp on windows from sources according to the procedure on http://yarp.it/git-master/install_yarp_windows.html.

MrHuangWenjie, 2021-02-17:
> Is a simulator of iCub running? If not, I think that that outcome is expected. Furthermore, how did you installed YARP? On which operating system?

I thought yarp as an iCub simulator platform tool. Do I need to install iCub separately?

traversaro, 2021-02-17:
Yes, `yarpmotorgui` is just a GUI to control the setpoints of joint-level control loops for either real or simulated robots, but just by launching yarpmotorgui you don't automatically spawn an iCub simulator.

There are two iCub simulations tools, `iCub_SIM`  and the Gazebo-based iCub simulation. Both require piece of software that is not part of YARP. I suggest to read the documentation on how to install iCub software in https://icub-tech-iit.github.io/documentation/sw_installation/ .

MrHuangWenjie, 2021-02-17:
> Yes, `yarpmotorgui` is just a GUI to control the setpoints of joint-level control loops for either real or simulated robots, but just by launching yarpmotorgui you don't automatically spawn an iCub [...]

Alright, thank you so much for the help!

traversaro, 2021-02-17:
I will close the issue, feel free to open other one if you have any specific problem, thanks!

closed 2021-02-17 by traversaro