robotology/yarp #2501: brew install robotology/formulae/yarp won't work on macbook pro with catalina 10.15.7

lucarag, 2021-02-19:
Hi, 
I followed the instructions listed both here:
http://www.yarp.it/git-master/install_yarp_mac.html#install_mac_Homebrew_YARP

and in this github repository. However, these instruction didn't work, with error messages below:

(base) C-LOSX0UEP3Y2:~ u79992lr$ brew tap homebrew/science
Error: homebrew/science was deprecated. This tap is now empty and all its contents were either deleted or migrated.
(base) C-LOSX0UEP3Y2:~ u79992lr$ brew install robotology/formulae/yarp
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
Error: No available formula or cask with the name "robotology/formulae/yarp".
==> Searching for a previously deleted formula (in the last month)...
Error: No previously deleted formula found.

I'm particularly interested in using yarp in python, and I actually also found this:
https://pypi.org/project/yarp/
but I'm unsure if this is just a wrapper and requires the above instructions to work first. 

Thank you in advance.

(I'm attaching the screenshot with error and instructions)
[image]

Nicogene, 2021-02-19:
Hi @lucarag,

The `yarp` formula has been removed since it was not maintained (see https://github.com/robotology/homebrew-formulae/pull/29)

I advise you to follow the instructions for [installing from sources](http://www.yarp.it/git-master/install_yarp_mac.html#install_mac_from_source).

Probably we should remove those instructions from the documentation cc @drdanz @traversaro

lucarag, 2021-02-22:
Hi @Nicogene,

I followed your advice and was able to complete the installation. I didn't have the chance to use the tool yet, but by typing yarp in the consolle, the output is the same as that in the guide. 

Thanke you again for you help!

closed 2021-02-23 by Nicogene