robotology/yarp #2512: Generate YARP_robotinterfaceConfig.cmake does not work if YARP_USE_SYSTEM_TinyXML is OFF

traversaro, 2021-02-25:
**Describe the bug**
If YARP_USE_SYSTEM_TinyXML  is OFF, the YARP_robotinterfaceConfig.cmake is not working properly.

**To Reproduce**
Not sure as this was problem was catched by @Iaxama, I think he just compiled YARP in a vanilla Ubuntu 20.04 image without `libtinyxml-dev` installed, and then he tried to compile gazebo-yarp-plugins 3.6.0 . 

**Expected behavior**
GazeboYARPPlugins should compile fine.

**Screenshots**
Error:
~~~
CMake Error at /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  Could not find a package configuration file provided by "YARP_priv_TinyXML"
  with any of the following names:

 

    YARP_priv_TinyXMLConfig.cmake
    yarp_priv_tinyxml-config.cmake

 

  Add the installation prefix of "YARP_priv_TinyXML" to CMAKE_PREFIX_PATH or
  set "YARP_priv_TinyXML_DIR" to a directory containing one of the above
  files.  If "YARP_priv_TinyXML" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  /usr/local/lib/cmake/YARP_robotinterface/YARP_robotinterfaceConfig.cmake:17 (find_dependency)
  /usr/local/lib/cmake/YARP/YARPConfig.cmake:162 (find_package)
  CMakeLists.txt:48 (find_package)
~~~

**Configuration (please complete the following information):**
 - OS: 20.04
 - yarp version: 3.4.3
-  compiler: Not relevant

**Additional context**
Add any other context about the problem here.

drdanz, 2021-02-25:
TinyXML (like the other "priv" libraries) is not installed when YARP is built as shared libraries, which was not an issue when robotinterface was just an executable, but it is now an issue.

I'm not sure how to handle this, because linking the static library to a shared library does not link it inside the shared library, but modifies the INTERFACE_LINK_LIBRARIES property.

Perhaps we can try making it an OBJECT library? But we should be very careful that the symbols from TinyXML are not exported...

traversaro, 2021-02-26:
There is any specific use case for which we want to compile and install `libYARP_robotinterface` without relying on a tinyxml dependency provided on the system package manager? If we want to support a compilation with zero external dependencies, we can simply have an option to disable either compilation or installation of `libYARP_robotinterface`, that should be sufficient to avoid the problem on this issue.

traversaro, 2021-02-26:
To clarify a bit the original issue, the only reason why we discovered this is that @laxama was building YARP and gazebo-yarp-plugins in a vanilla docker, and there he did not installed `libtinyxml-dev`, and this triggered the problem. All other users I am aware of tipically just install YARP via the robotology-superbuild, and there tinyxml is always installed via apt/homebrew/vcpkg/conda .

drdanz, 2021-03-25:
Sorry, I missed the replies... I don't know about disabling `libYARP_robotinterface`, it is becoming a pretty important component...
I think we should just re-discuss the whole "zero external dependencies" thing...

traversaro, 2021-03-25:
>  I think we should just re-discuss the whole "zero external dependencies" thing...

I agree, with the rise of user-space package managers such as vcpkg or conda even for C++ and for Windows (where package managers have been historically lacking), I think the need for hard zero external dependencies is less strong. Even if you need to cross-compile everything for an esoteric platform (WebAssembly? Embedded RTOS?), setting up a superbuild or a colcon workspace of all the required libraries and compile all of them with the correct CMake toolchain is not difficult considering all the available tooling.

PeterBowman, 2022-07-16:
This has just happened to me on a fresh Ubuntu 20.04 install with YARP version 3.7.2+1-20220630.8+gitc8450ac9e while building latest gazebo-yarp-plugins (d1995eac52adf6a273185da3eab2f279718725b9):

```
CMake Error at /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  Could not find a package configuration file provided by "YARP_priv_TinyXML"
  with any of the following names:

    YARP_priv_TinyXMLConfig.cmake
    yarp_priv_tinyxml-config.cmake

  Add the installation prefix of "YARP_priv_TinyXML" to CMAKE_PREFIX_PATH or
  set "YARP_priv_TinyXML_DIR" to a directory containing one of the above
  files.  If "YARP_priv_TinyXML" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  /usr/local/lib/cmake/YARP_robotinterface/YARP_robotinterfaceConfig.cmake:17 (find_dependency)
  /usr/local/lib/cmake/YARP/YARPConfig.cmake:159 (find_package)
  CMakeLists.txt:49 (find_package)
```

Oddly enough, it actually configures successfully when I replace `YARP_priv_TinyXML` with `YARP_priv_tinyxml`, which seems to be the correct component package name. Compare:

https://github.com/robotology/yarp/blob/05ca89400b36bf26aea8cb4707f3029d14ccbda7/extern/tinyxml/CMakeLists.txt#L43

versus:

https://github.com/robotology/yarp/blob/05ca89400b36bf26aea8cb4707f3029d14ccbda7/src/libYARP_robotinterface/src/CMakeLists.txt#L105

I see more components depending on _TinyXML (such as YARP_manager) instead of _tinyxml.

PeterBowman, 2022-07-16:
For some reason, `BUILD_SHARED_LIBS` set itself to `OFF` for me, hence [InstallBasicPackageFiles](https://github.com/robotology/ycm/blob/2583228ed11f0fe5ba32497943b112099cd1af81/modules/InstallBasicPackageFiles.cmake#L598-L619) interpreted my installation as static and requested the TinyXML component. Apparently, I also had `YARP_BUILD_TinyXML` set to `ON`. I ran a CMake configure again and both variables toggled. I wouldn't give myself too much credit here...

In any case, I think the component naming issue is a bug indeed. To reproduce, configure with `-DBUILD_SHARED_LIBS=OFF -DYARP_USE_SYSTEM_TinyXML=OFF`. This generates a *build/YARP_robotinterface/YARP_robotinterfaceConfig.cmake* file with the following line:

```cmake
find_dependency(YARP_priv_TinyXML HINTS "${CMAKE_CURRENT_LIST_DIR}/.." NO_DEFAULT_PATH)
```

Yet the actual component (as can be seen in *build/*) is *YARP_priv_tinyxml*.

jgvictores, 2023-05-27:
This has just happened to me on a Ubuntu 20.04.3 install with YARP version 3.8.0 while building latest gazebo-yarp-plugins (e613f5296c07675f8bf946299e5b1ed5e787bedf).

traversaro, 2025-03-21:
Fix suggested in https://github.com/robotology/yarp/pull/3199 .