robotology/yarp #2513: Syntax error in YARP examples

traversaro, 2021-02-28:
**Describe the bug**
While packaging YARP for conda in https://github.com/traversaro/robotology-superbuild/runs/1995112364?check_suite_focus=true, I am getting this warning for the YARP python examples:
~~~
2021-02-27T19:27:23.5916178Z Resource usage statistics from building yarp:
2021-02-27T19:27:23.5916698Z    Process count: 10
2021-02-27T19:27:23.5917130Z    CPU time: Sys=0:00:25.9, User=0:07:26.8
2021-02-27T19:27:23.5917521Z    Memory: 720.6M
2021-02-27T19:27:23.5917894Z    Disk usage: 302.8K
2021-02-27T19:27:23.5918282Z    Time elapsed: 0:16:09.8
2021-02-27T19:27:23.5918556Z 
2021-02-27T19:27:23.5918831Z 
2021-02-27T19:27:23.5919168Z Packaging yarp
2021-02-27T19:27:33.3757280Z   File "share/yarp/bindings/python/examples/example_enc.py", line 26
2021-02-27T19:27:33.3759184Z     print 'Opening the motor driver...'
2021-02-27T19:27:33.3759943Z           ^
2021-02-27T19:27:33.3761161Z SyntaxError: Missing parentheses in call to 'print'. Did you mean print('Opening the motor driver...')?
2021-02-27T19:27:33.3762038Z 
2021-02-27T19:27:33.3765026Z   File "share/yarp/bindings/python/examples/yarpImage_py2.py", line 45
2021-02-27T19:27:33.3766389Z     print 'Receiving image from input port...'
2021-02-27T19:27:33.3767147Z           ^
2021-02-27T19:27:33.3770237Z SyntaxError: Missing parentheses in call to 'print'. Did you mean print('Receiving image from input port...')?
2021-02-27T19:27:33.3771242Z 
2021-02-27T19:27:34.6663350Z compiling .pyc files...
~~~

**To Reproduce**
Inspect the files and check the syntax

**Expected behavior**
The examples should work fine in Python3 

**Screenshots**

**Configuration (please complete the following information):**
 - OS: Ubuntu 20.04 with conda-forge dependencies
 - yarp version: 3.4.3
-  compiler: Not Relevant

traversaro, 2021-02-28:
I guess that it should be sufficient to just add the parenthesis to the print, but perhaps Python-savvy users can spot more errors or try the examples directly. 

fyi @xEnVrE @GiulioRomualdi @prashanthr05 @diegoferigo @paolo-viceconte

diegoferigo, 2021-03-01:
For completeness, the file is stored in this folder: https://github.com/robotology/yarp/tree/master/bindings/python/examples.

It look quite good to me, just a nit about `Network`. Maybe we can avoid to explicitly call `fini()` if we instantiate a network object instead of initializing it from the static method. It gets finalized when the objects gets destroyed. Note that in complex Python application, this is quite delicate because objects are destroyed not when they go out of scope but when the garbage collector decides to free some resources. In this specific case is not a big problem since we're taking of the network object, but it's something Python users should keep in mind.

jgvictores, 2021-03-01:
> I guess that it should be sufficient to just add the parenthesis to the print, but perhaps Python-savvy users can spot more errors or try the examples directly.

It's one of the main differences between Python 2 and Python3 (b/w compat for latest Python 2 versions, obsolete anyway). A relatively safe way for migration is via [2to3](https://docs.python.org/3/library/2to3.html), which is official and available in most distros.

PeterBowman, 2021-07-13:
Examples have been updated in the meantime and most of them run nicely. The [yarpImage.py](https://github.com/robotology/yarp/blob/88f97f3c80a975dcfd2b46497a713b69f37b548f/bindings/python/examples/yarpImage.py) one renders segfaults for me in Python 3.8.10, looks like it's https://github.com/robotology/yarp/issues/1873 again. And while there are no syntax errors in [example_callback.py](https://github.com/robotology/yarp/blob/88f97f3c80a975dcfd2b46497a713b69f37b548f/bindings/python/examples/example_callback.py), it does nothing since I forgot to add a `p.open('/python')` after this line:

https://github.com/robotology/yarp/blob/88f97f3c80a975dcfd2b46497a713b69f37b548f/bindings/python/examples/example_callback.py#L30