robotology/yarp #2521: Error: cannot connect to /root port; port port_name failed to activate (invalid address)

Gennaro91, 2021-03-18:
Hello everyone,
I'm a beginner in using YARP. My goal is to create two ports, and connect them (I'm using an Ubuntu virtual machine). The writer should receive inputs from the command line through the argv pointer.

When I run the code:

```
#include <yarp/os/all.h>
#include <iostream>

using namespace yarp::os;

int main(int argc, char *argv[]) {
int i = 0;

    // Set up YARP
    Network yarp;

    // Create port
    Port pout;
    bool ok  = pout.open("/examples/port/sender");
    if (!ok) {
        std::cerr << "Failed to open port" << std::endl;
        return 1;
    }

    // waiting for the receiver before transmitting
    yarp.waitConnection("/examples/port/sender", "/examples/port/receiver");

    // send data in a loop
    for(i = 0; i < argc; i++) {
        Bottle b;
    	int x = atoi(argv[i]);
    	b.addInt(x);
    	std::cout << "Sending " << x << " message " << std::endl;
        pout.write(b);
    }

    // close port
    pout.close();

    return 0;
}
```
I got the following errors:
```
[ERROR] |yarp.os.Network| Cannot connect to port /root
[ERROR] |yarp.os.Port| Port /examples/port/sender failed to activate (invalid address)
Failed to open port
```

Can you tell me what is the mistake I did please?

Thank you for your time.

PeterBowman, 2021-03-18:
Hello. The `/root` port refers to the YARP name server in the default namespace. Please make sure you have started it, e.g. via `yarp server`, before running your application. It might require you to add the `--write` option. I have tested your code with YARP server down and got the exact same error. Once you start it and run the application again, you should see:

```
[INFO] |yarp.os.Port| Port /examples/port/sender active at tcp://192.168.1.53:10003/
[INFO] |yarp.os.Network| Waiting for /examples/port/sender->/examples/port/receiver...
```

You can also check the availability of the name server in C++ code via [`yarp::os::Network::checkNetwork()`](http://www.yarp.it/git-master/classyarp_1_1os_1_1NameSpace.html#a034c4d3159d0af2aa09083a1b16f411d).

drdanz, 2021-03-18:
I think I already replied on [StackOverflow](https://stackoverflow.com/questions/66690911) but I'll paste the reply here as well... 😄 

---

Both errors come from the same issue:

    [ERROR] |yarp.os.Network| Cannot connect to port /root

This means that the `/root` is not available when you run the executable. The `/root` port is the port used by `yarp server` (Unless you change the default "namespace").

To fix this error you should just run `yarp server` on a different shell.

    [ERROR] |yarp.os.Port| Port /examples/port/sender failed to activate (invalid address)

The `/root` port cannot be contacted, and therefore YARP is trying to open a port on an invalid address. It should be fixed as well as soon as you run the executable while the `yarp server` is running.

You can find a very basic tutorial about ports here: http://www.yarp.it/git-master/note_ports.html

closed 2021-03-18 by drdanz