robotology/yarp #2535: yarpdatadumper with png files

triccyx, 2021-03-29:
**Describe the bug**
Using yarpdatadumper can not save images in png format. While bmp format works (--image)

**To Reproduce**
```bash
yarpserver --write
yarpdev --device grabberDual --subdevice usbCamera --camModel ultrapython --d /dev/media0 --name /grabber
yarpdatadumper --name /log --rxTime --txTime --type image_png --connect /grabber
```

**Expected behavior**
Inside folder log I can find the dumped images. No images found.

**Screenshots**
na

**Configuration (please complete the following information):**
 - OS:Ubuntu 20.04
 - yarp version:YARP version 3.4.101+20210309.4+gitae65e3050+dirty
-  compiler:gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0

**Additional context**
Errors issued repeatedly:
```
libpng warning: Invalid bit depth in IHDR
libpng error: Invalid IHDR data
[ERROR]/home/triccyx/Documents/robotology-superbuild/robotology/YARP/src/libYARP_sig/src/yarp/sig/ImageFile.cpp:316 bool SavePNG(char*, const char*, size_t, size_t, size_t, png_byte, png_byte): [write_png_file] Error during writing header
libpng warning: Invalid bit depth in IHDR
libpng error: Invalid IHDR data
[ERROR]/home/triccyx/Documents/robotology-superbuild/robotology/YARP/src/libYARP_sig/src/yarp/sig/ImageFile.cpp:316 bool SavePNG(char*, const char*, size_t, size_t, size_t, png_byte, png_byte): [write_png_file] Error during writing header
```
Important notes:
- Try with the new UltraPython camera
- yarpdev on ARM arch, yarpdatadumper on X86 or Arm is the same result

pattacini, 2021-03-29:
By launching:
```console
$ yarpdev --device fakeFrameGrabber &
$ yarpdatadumper --connect /grabber --type image_png
```
I managed to get the PNG image stored correctly.

Could you do the same test on your system @triccyx.

triccyx, 2021-03-29:
I have tried on the ARM Enclustra board and all seems ok. The files are generated.

closed 2021-03-29 by Nicogene