robotology/yarp #2536: Unable to launch YARP network on yarp@master

pattacini, 2021-03-29:
**Describe the bug**
I'm unable to launch the YARP network using `yarp@master` on Windows.
Everything works as expected with `yarp@v3.4.3`.

**To Reproduce**
Steps to reproduce the behavior:
- `yarpserver`
- `yarp where` --> no detection of the server
- `yarp detect` --> server correctly detected

**Expected behavior**
`yarp where` shall detect the server properly.

**Configuration (please complete the following information):**
 - OS: Windows 10 Pro 20H2 19042.870
 - yarp version: `master`
-  compiler:
  Microsoft Visual Studio Enterprise 2019
  Version 16.9.2
  VisualStudio.16.Release/16.9.2+31112.23
  Microsoft .NET Framework
  Version 4.8.04084
  Installed Version: Enterprise
  Visual C++ 2019   00434-00052-67056-AA269
  Microsoft Visual C++ 2019

**Additional context**
Add any other context about the problem here.

Interestingly enough, I'm getting the following error message while issuing `yarp conf [IP] [port]` correctly:
```console
Configuration file:
  C:\Users\upattacini\AppData\Roaming\yarp\config\_ugo.conf
[ERROR] is not valid!
[ERROR] Expected:
[ERROR]   yarp conf [ip address] [port number]
[ERROR]   yarp conf [ip address] [port number] [yarp|ros]
[ERROR] For example:
[ERROR]   yarp conf 192.168.0.1 10000
```

pattacini, 2021-03-29:
cc @traversaro

traversaro, 2021-03-29:
I double checked in my Windows machine with conda, and 3.4.3 works fine:
~~~
conda create -n yarp2536
conda activate yarp2536
conda install -c conda-forge -c robotology yarp
~~~
and this is the output with `yarp conf`
~~~
(yarptest)
STraversaro@IITICUBLAP102 MINGW64 ~
$ yarp conf 0.0.0.0 1234
Configuration file:
  C:\Users\STraversaro\AppData\Roaming\yarp\config\yarp.conf
Stored:
  host 192.168.1.200 port number 10000 (yarp name server)
Now stores:
  host 0.0.0.0 port number 1234 (yarp name server)
(yarptest)
~~~

drdanz, 2021-03-30:
There was a regression in `yarp conf` that I fixed a couple of minutes ago in master, please check now and let me know if that was fixed, or if you are having some other issue with multicast detection or similar...

pattacini, 2021-03-30:
Tested it right now and it's working correctly.
Thanks 👍🏻

closed 2021-03-30 by pattacini