robotology/yarp #2544: Change YARP_PLUM_BOB to YARP_PLUMB_BOB

ste93, 2021-04-08:
I've noticed that PLUMB_BOB distortion is badly written in the code, we can update it to the right one to avoid incoerences.

drdanz, 2021-04-13:
We need to pay attention since the `enum class YarpDistortion` is part of the public API.
If we want to do it clean, we should keep both names, declaring one deprecated...
Anyway, it seems to be used only in YARP and  yarp-device-realsense2, as long as they are kept in sync, I think there should not be a problem in breaking the API

ste93, 2021-04-14:
I've also opened an issue in yarp-device-realsense2 https://github.com/robotology/yarp-device-realsense2/issues/10

closed 2021-07-02 by drdanz

traversaro, 2022-04-05:
For reference, this was fixed by https://github.com/robotology/yarp/pull/2593 that was eventually released as part of YARP 3.5 .