robotology/yarp #2558: yarpdataplayer timestamp gets rounded because of Ubuntu regional format

aikolina, 2021-04-27:
**Describe the bug**
When checking the envelope of the data coming out of `yarpdataplayer`, the timestamp gets rounded to an integer.

**To Reproduce**
1. Ubuntu regional format should be _Italian_ (or any other language using `,` instead of `.` for decimals). 
2. Load some data in the `yarpdataplayer` and enable the _Strict_ option.
3. Read the `yarpdataplayer` port from the command line using `yarp read ... /port_name envelope`.
4. Check the timestamp format.

**Expected behavior**
For example, if the considered input data is:
```
1 0.003000 AE (37499 1651768 ...
```
Then, the output of `yarp read ... /port_name envelope` should be:
```
1 0.00300000000000000006245 AE (37499 1651768 ...
```
Instead, I was getting:
```
1 0 AE (37499 1651768 ...
```

**Configuration:**
 - OS: Ubuntu 18.04.5 LTS
 - yarp version: 3.4.3+38-20210330.6+gitf7f000f39
-  compiler: not relevant

**Solution**
It was a problem of the regional format in Ubuntu: `yarpdataplayer` interpreted the timestamp correctly after I changed from _Italian_ to _English (United States)_  and restarted my laptop.

[image]

drdanz, 2021-04-28:
`yarp::conf::{from_string,to_string}` and `yarp::os::NetType::{fromString,toString}` are supposed to take care of that, see

https://github.com/robotology/yarp/blob/493d7a390d693b82b1ea2e8feecce3b03c478386/src/libYARP_conf/src/yarp/conf/numeric.h.in#L144-L156

https://github.com/robotology/yarp/blob/493d7a390d693b82b1ea2e8feecce3b03c478386/src/libYARP_conf/src/yarp/conf/numeric.h.in#L293-L300

I believe that there might be some place in `yarpdataplayer` where these functions are not used, and the data is read "raw"...

drdanz, 2021-04-28:
CC @vtikha @vvasco

vvasco, 2021-04-29:
A PR has been opened [here](https://github.com/robotology/yarp/pull/2560) to fix the issue.

closed 2021-05-03 by drdanz