robotology/yarp #2568: v3.4.4 patch release

traversaro, 2021-05-13:
**Is your feature request related to a problem? Please describe.**
The `yarp-3.4`  branch contains some useful Python-related fixes (see https://github.com/robotology/yarp/pull/2527) that it would be convenient to have in a released version of YARP, to support the generation of Python bindings binaries, see https://github.com/robotology/robotology-superbuild/issues/641 .

**Describe the solution you'd like**
Tag the `v3.4.4` release.

**Describe alternatives you've considered**
Using a non-tagged checkout out of the `yarp-3.4` branch, or add support to apply patches at the robotology-superbuild level.

**Additional context**
This is convenient to have a properly tagged version of YARP for robotology-superbuild version 2021.05 . This is similar to https://github.com/robotology/yarp/issues/2458 .

traversaro, 2021-05-13:
cc @Nicogene

drdanz, 2021-05-19:
YARP 3.4.4 released.

closed 2021-05-19 by drdanz

traversaro, 2021-05-19:
Thanks @drdanz !