robotology/yarp #2586: Add methods to retrieve names of entities in yarp::dev::IJoypadController interfaces?

traversaro, 2021-06-01:
**Is your feature request related to a problem? Please describe.**
At the moment, the ` yarp::dev::IJoypadController` identifies joypad entities such as Axis, Button, Trackball, etc, etc only with numerical IDs. This mimicks the SDL2 APIs that I guess was the original inspiration for `yarp::dev::IJoypadController`. 
However, with @S-Dafarra we are starting to investigate how to expose via the `yarp::dev::IJoypadController` interface and the related devices joypads that can be accessed via the [OpenXR API](https://www.khronos.org/openxr/).

OpenXR supports the identification of joypad entities via strings, see (https://www.khronos.org/registry/OpenXR/specs/1.0/html/xrspec.html#semantic-path and https://www.khronos.org/registry/OpenXR/specs/1.0/html/xrspec.html#semantic-path-interaction-profiles).

For this reason, it would be convenient to have some methods to expose this information that OpenXR exposes.

**Describe the solution you'd like**
A possible solution is to add the following methods to `yarp::dev::IJoypadController`
~~~cxx
virtual bool 	getAxisName(unsigned int axis_id,  std::string &name);
virtual bool 	getButtonName(unsigned int button_id,  std::string &name);
virtual bool 	getTrackballName(unsigned int trackball_id,  std::string &name);
virtual bool 	getHatName(unsigned int hat_id,  std::string &name);
virtual bool 	getTouchSurfaceName(unsigned int touch_id, std::string &name);
virtual bool 	getStickCount(unsigned int stick_it, std::string &name);
~~~
with default implementations that return false. At the `JoypadControlServer`/`JoypadControlClient` level, this info can be exposed via RPC port.

**Describe alternatives you've considered**
Possible alternatives are:
* Add those methods without default implementation, breaking compatibility with devices not in YARP that inherit from  `yarp::dev::IJoypadController` .
* Create a  yarp::dev::IJoypadController2` interface with these additional methods.

drdanz, 2021-06-03:
I don't know if this makes sense, but since not all controllers have these information, another option, instead of adding it to the `IJoypadController` interface, could be to add a new `IJoypadControllerInfo` interface.

(P.S. `getStickCount` above, should probably be `getStickName`)

traversaro, 2021-06-03:
> 
> 
> I don't know if this makes sense, but since not all controllers have these information, another option, instead of adding it to the `IJoypadController` interface, could be to add a new `IJoypadControllerInfo` interface.
> 

Good point, but I guess it could still make sense to reuse the existing `JoypadControlServer`/`JoypadControlClient` devices, right?

> (P.S. `getStickCount` above, should probably be `getStickName`)

Thanks, fixed!

drdanz, 2021-06-03:
> Good point, but I guess it could still make sense to reuse the existing `JoypadControlServer`/`JoypadControlClient` devices, right?

Yes, these devices should be updated to support the new interface...