robotology/yarp #2620: portaudioPlayer  regression in YARP master branch (upcoming 3.5 release)

traversaro, 2021-06-21:
**Describe the bug**
On some systems, including:
* `conga-TC87/i7-4650U`  (CPU used in iCub3, see https://icub-tech-iit.github.io/documentation/icub_cpu_boards/icub_cpu_boards/) with Ubuntu 18.04 

The `portaudioPlayer` YARP device is not working as expected on YARP master branch. In particular, even if on the system the audio can be played correctly using PortAudio examples (see https://github.com/PortAudio/portaudio/tree/pa_stable_v190600_20161030/examples), the `portaudioPlayer` YARP devices is not able to playback audio. On the same system, downgrading YARP to the `v3.4.5` version fixed the issue, permitting to reproduce audio via the `portaudioPlayer` YARP device. 

**To Reproduce**
In our case, the problem appeared on the iCubGenova09 system, composed of `icub-laptop`  and `icub-head` .  The test done is described in the following.

The  `yarpdev --device AudioPlayerWrapper --subdevice portaudioPlayer --name /icub/speakers --playback_network_buffer_lenght 0.4` command was not resulting in any audio being played, while the `./paex_sine` command is able to play a sound. So we used the `yarpdev --device fakeMicrophone` command to generate some audio, and try to play it back on the speakers by using the following commands:
~~~
yarp rpc /audioRecorderWrapper/rpc (and then type start)
yarp connect /audioRecorderWrapper/audio:o /icub/speakers/audio:i
~~~

We tried to do so with this different combinations:

| `fakeMicrophone` running on? | `portaudioPlayer` running on?  |   Outcome |
|:---------------------------------:|:----------------------------------:|:------------:|
|  `icub-head`                             |  `icub-head`                                |  Still needs to be tested  ❓ |
|  `icub-console`                             |  `icub-head`                                |  No sound is  played ❌ |
|  `icub-head`                             |  `icub-console`                                |  Sound is played on laptop speakers  ✅ |
|  `icub-console`                             |  `icub-console`                                |  Sound is played on laptop speakers ✅ |

This seems to indicate that indeed the `portaudioPlayer` running on `icub-head` has some problems. 

However, note that the version of YARP are on different commits on icub-head and icub-console, so the `yarpdev --device fakeMicrophone` when executed on `icub-console` was a bit different:
`icub-head`: https://github.com/robotology/yarp/commit/887699bba1c03d6edc5ea6050791f757bced8342 
`icub-console` : https://github.com/robotology/yarp/commit/7d375628f352dc27f207830cb80a4179c4617a54 

Later, @isorrentino experience the same problem on `icub-head`, and solved the problem by downgrading YARP there to `v3.4.5` .

**Expected behavior**
`yarpdev --device AudioPlayerWrapper --subdevice portaudioPlayer --name /icub/speakers --playback_network_buffer_lenght 0.4` should be able to play back sound from `yarpdev --device fakeMicrophone` both on YARP v3.4.5 and the latest YARP master.

**Screenshots**
N.A. 

**Configuration (please complete the following information):**
 - OS: Ubuntu 18.04
 - yarp version: https://github.com/robotology/yarp/commit/887699bba1c03d6edc5ea6050791f757bced8342 
-  compiler: Ubuntu 18.04 default compiler

**Additional context**
@isorrentino if you have any additional info please include it, thanks!

traversaro, 2021-06-21:
Not sure if this is related to https://github.com/robotology/yarp/pull/2433 or if on some other systems the `portaudioPlayer`  devices is able to run correctly even using one of the last commits in YARP master branch.

pattacini, 2021-06-21:
cc @davidetome @vtikha

drdanz, 2021-06-21:
cc @randaz81

S-Dafarra, 2021-07-02:
After a F2F discussion with @randaz81, it turned out that it might be a simple configuration issue. In particular, with https://github.com/robotology/yarp/commit/e5c8682c6f09554b71e833e7698fff1a3816070b the audio player will not start automatically anymore, unless a ``--start`` option is added: https://github.com/robotology/yarp/blob/3c1d4fbc0eaa576d165e1461cc6448efe968a347/src/devices/audioPlayerWrapper/AudioPlayerWrapper.cpp#L264

For future references, it is also a good move to check the output of the ``status:o`` port, which also includes information on whether the device is running or not, together with the dimension of the buffer: https://github.com/robotology/yarp/blob/3c1d4fbc0eaa576d165e1461cc6448efe968a347/src/devices/audioPlayerWrapper/AudioPlayerWrapper.cpp#L351-L353

These devices were still in active development before ``3.5``, and with this release, they should also be properly documented.

I just tried on my laptop (the robot where we had the issue is under maintenance) and by simply adding the ``--start`` flag I am able to run the test mentioned in https://github.com/robotology/yarp/issues/2620#issue-926229104

Apologies if I missed something.

cc @GiulioRomualdi @isorrentino

traversaro, 2021-07-02:
Great! Do you think we can then close the issue?

S-Dafarra, 2021-07-02:
I think so. In case of problems we can reopen it I think

closed 2021-07-02 by traversaro