robotology/yarp #2630: v3.4.3 regression: yarprobotinterface floods its output with `Command received` and `Response sent:` messages

traversaro, 2021-07-01:
**Describe the bug**
After https://github.com/robotology/yarp/pull/2485/files, yarprobotinterface on the `yarp-3.4` branch and on releases v3.4.3, v3.4.4 and v3.4.5 is flooding its output with `Command received` and `Response sent:`, making really hard to read the other relevant output. The problem was fixed in master (YARP 3.5) in https://github.com/robotology/yarp/commit/d02db0521aae86537aa5ada8bc17ce661d2cb3e3#diff-3bbd8f68798112a6c309136664e4e9bb42af65bec13be7143219a14551c38aa8 , but the nature of the bug is such that it is complicated to use yarprobotinterface with v3.4.3/v3.4.4/v3.4.5 releases.

**To Reproduce**
Run `yarprobotinterface` (probably @xEnVrE @S-Dafarra can add more input).

**Expected behavior**
The log should contains relevant output, not the same message repeatdly printed. 

**Screenshots**

**Configuration (please complete the following information):**
 - OS: Ubuntu 20.04
 - yarp version: v3.4.3
-  compiler: GCC 9

**Additional context**
This problem was detected by both @S-Dafarra and @xEnVrE .

traversaro, 2021-07-01:
A fix for this issue is provided in https://github.com/robotology/yarp/pull/2631 .

traversaro, 2021-07-01:
@S-Dafarra @xEnVrE as a temporary workaround, you can ensure that YARP does not print `Debug` statements by defining the environment variable `YARP_DEBUG_ENABLE` to `0` , see http://www.yarp.it/git-master/yarp_logging.html#autotoc_md911 for more info.

S-Dafarra, 2021-07-01:
Thanks a lot @traversaro for opening the issue. While rushing for a deadline, we simply commented out the lines affected by https://github.com/robotology/yarp/commit/d02db0521aae86537aa5ada8bc17ce661d2cb3e3

S-Dafarra, 2021-07-01:
> Run `yarprobotinterface` (probably @xEnVrE @S-Dafarra can add more input).

That was happening when opening a ``yarpmotorgui``, saturating the logger in seconds.

closed 2021-07-02 by drdanz