robotology/yarp #2640: Python bindings: issues with clock initialization under windows

Nicogene, 2021-07-05:
**Describe the bug**
I am experiencing problems opening `Polydriver`s under windows using the python bindings.
Seems that `yarp.Network.init()` and `yarp.useSystemClock` does not work.
(I am using python 3.7.10).
I am getting this error while trying to open `test_grabber`:
```
[ERROR] |yarp.os.Time| Warning an issue has been found, please update the code.
[ERROR] |yarp.os.Time|  This means YARP framework has not been properly initialized.
[ERROR] |yarp.os.Time|  The clock can be initialized with one of the following methods:
[ERROR] |yarp.os.Time|  - Create yarp::os::Network object or call yarp::os::Network::init()
[ERROR] |yarp.os.Time|  - Call useSystemClock()
[ERROR] |yarp.os.Time|  otherwise use yarp::os::SystemClock::nowSystem() and yarp::os::SystemClock::delaySystem() instead of Time::now() and Time::delay()
```

On WSL2 (Ubuntu 20.04.2) I did not have any issues with it.

**To Reproduce**
Simply run this script under windows:
```python
import yarp
yarp.Network.init()
options = yarp.Property()
driver = yarp.PolyDriver()
options.put("device","test_grabber")
driver.open(options)
```

**Expected behavior**
The driver of `test_grabber` should be opened also under windows

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Configuration (please complete the following information):**
 - OS: Windows 10, version 21H1
 - yarp version: 3.4.5
-  compiler: VS 2019

**Additional context**
Add any other context about the problem here.

traversaro, 2021-09-08:
How do you installed the YARP bindings? If with conda, can you include the exact packages used with `conda list`?

Nicogene, 2021-09-08:
I have installed them with conda indeed, here is my `conda list`:

```
# packages in environment at C:\Users\ngenesio\AppData\Local\miniforge3\envs\blenderenv:
#
# Name                    Version                   Build  Channel
ace                       7.0.2                h0e60522_0    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.5.30            h5b45459_0    conda-forge
certifi                   2021.5.30        py37h03978a9_0    conda-forge
freeglut                  3.2.1                h0e60522_2    conda-forge
gsl                       2.6                  hdfb1a43_2    conda-forge
icu                       68.1                 h0e60522_0    conda-forge
icub-firmware-shared      1.20.0              h0e60522_13    robotology
icub-main                 1.20.0              h9e6d7c3_13    robotology
idyntree                  3.2.0           py37h542c6b5_13    robotology
intel-openmp              2021.2.0           h57928b3_616    conda-forge
ipopt                     3.13.4               hf6be2e5_0    conda-forge
irrlicht                  1.8.4                h739eaf8_1    conda-forge
jpeg                      9d                   h8ffe710_0    conda-forge
libblas                   3.9.0                     9_mkl    conda-forge
libcblas                  3.9.0                     9_mkl    conda-forge
libclang                  11.1.0          default_h5c34c98_1    conda-forge
libflang                  5.0.0           h6538335_20180525    conda-forge
libiconv                  1.16                 he774522_0    conda-forge
libjpeg-turbo             2.1.0                h8ffe710_0    conda-forge
liblapack                 3.9.0                     9_mkl    conda-forge
libosqp                   0.6.2                h0e60522_3    conda-forge
libpng                    1.6.37               h1d00b33_2    conda-forge
libxml2                   2.9.12               hf5bbc77_0    conda-forge
llvm-meta                 5.0.0                         0    conda-forge
metis                     5.1.0             h0e60522_1006    conda-forge
mkl                       2021.2.0           hb70f87d_389    conda-forge
mumps-seq                 5.2.1               hb3f9cae_10    conda-forge
numpy                     1.20.3           py37hcbcd69c_1    conda-forge
openmp                    5.0.0                    vc14_1    conda-forge
openssl                   1.1.1k               h8ffe710_0    conda-forge
osqp-eigen                0.6.3               h0e60522_13    robotology
pip                       21.1.2             pyhd8ed1ab_0    conda-forge
python                    3.7.10          h7840368_100_cpython    conda-forge
python_abi                3.7                     1_cp37m    conda-forge
qt                        5.12.9               h5909a2a_4    conda-forge
robot-testing-framework   2.0.1               h0e60522_13    robotology
sdl                       1.2.15               h13ae965_1    conda-forge
setuptools                49.6.0           py37h03978a9_3    conda-forge
sqlite                    3.35.5               h8ffe710_0    conda-forge
tbb                       2021.2.0             h2d74725_0    conda-forge
tinyxml                   2.6.2                h2d74725_2    conda-forge
tk                        0.1.0                    pypi_0    pypi
vc                        14.2                 hb210afc_4    conda-forge
vs2015_runtime            14.28.29325          h5e1d092_4    conda-forge
wheel                     0.36.2             pyhd3deb0d_0    conda-forge
wincertstore              0.2             py37h03978a9_1006    conda-forge
yarp                      3.4.5           py37h20261aa_13    robotology
ycm-cmake-modules         0.12.2              h0e60522_13    robotology
zlib                      1.2.11            h62dcd97_1010    conda-forge
```

traversaro, 2021-09-08:
> On WSL2 (Ubuntu 20.04.2) I did not have any issues with it.

Did you used apt or conda in that case?

traversaro, 2021-09-08:
OT: Why did you installed https://pypi.org/project/tk/ via pip ?

Nicogene, 2021-09-08:
> Did you used apt or conda in that case?

If I remember correctly I followed this https://github.com/robotology/robotology-superbuild/blob/master/doc/install-miniforge.md#linux

> OT: Why did you installed https://pypi.org/project/tk/ via pip ?

Mm I don't even remember why I installed it sorry 😅

drdanz, 2021-09-08:
(Edited `test_grabber` -> `fakeFrameGrabber`)

I was able to run the example without the error using the python3-yarp package from here: https://launchpad.net/~robotology/+archive/ubuntu/test/

<details>
 <summary><em>Minimal steps to reproduce with Docker</em></summary>

`Dockerfile`
```Dockerfile
FROM ubuntu:focal
RUN apt-get update -qq && \
    DEBIAN_FRONTEND=noninteractive apt-get install -y software-properties-common && \
    add-apt-repository -y -u ppa:robotology/test && \
    apt-get update -qq && \
    DEBIAN_FRONTEND=noninteractive apt-get install -y yarp python3-yarp && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*
```

```bash
docker build -t my-yarp-apt:latest .
docker run -it --rm --network=host -v${HOME}/.config/yarp:/root/.config/yarp my-yarp-apt:latest bash
```
</details>

It seems to be partially working (weirdly the device is opened successfully, and the RPC port is ok, but the output port is not opened). I don't see your error though... It might indeed be an issue in the conda packaging (perhaps something related to the `YARP_init` library?)

Nicogene, 2021-09-08:
@drdanz with that docker you tested it under Linux, am I right?
On Linux I have no issues either with or without conda

drdanz, 2021-09-08:
Yes, sorry, I did not understand this was a windows issue...