robotology/yarp #2643: Yarplogger may not report all entries

sgiraz, 2021-07-06:
**Bug description**

I happened to spot an unexpected behavior concerning the tracing of operations such as `yInfo`, `yWarning`, `yError` etc. within the `yarplogger`.

If two operations are very close each other in time, I can observe that the `yarplogger` may not keep track of their invocations. In particular, the lower the time interval the higher the probability of a missed trace.

**To Reproduce**

In order to reproduce the behavior with the smallest code snippet, I crafted a [`yInfo-test`](https://github.com/sgiraz/yInfo-test) repository where you can find a simple case study (fully documented).

The [`yInfo-test`](https://github.com/sgiraz/yInfo-test) application contains a simple for-loop that performs 10k iterations with just one call to the `yInfo` function per cycle. Note that after each iteration a [yarp::os::Time::delay](http://www.yarp.it/git-master/namespaceyarp_1_1os_1_1Time.html#aca80809e0597551745954e2547e581ee) is also called so we can vary the elapsed time between the `yInfo` calls. 

The test scenario foresees the use of `yarplogger` without the `yarprun` broker, thus relying on the `YARP_FORWARD_LOG_ENABLE=1` env var. 

After some quick tests, it came out that there exists a threshold of 0.001s. Below this value (included), the number of missing `yInfo` logged by the `yarplogger` dramatically increases.

Please, notice that the for-loop is only aimed to replicate the behavior, whereas two very close logging operations may occur wherever in the user code. 

Below, you can see some results comparing what was produced by the `yarplogger` with the expected results: 

<details>
<summary>🔘 Click to see the result of Test 1</summary>

```console
./check_log_file.sh build/yarprunlog_04_07_2021_23_46_12.log
Parsing log file...
Missing value 1707, previous was at [ Time delay ]: 0.001 Counter value: 1706
Done.
```
</details>

<details>
<summary>🔘 Click to see the result of Test 2</summary>

```console
./check_log_file.sh build/yarprunlog_04_07_2021_23_41_24.log
Parsing log file...
Missing value 3224, previous was at [ Time delay ]: 0.0001 Counter value: 3223
Missing value 3512, previous was at [ Time delay ]: 0.0001 Counter value: 3511
Missing value 3513, previous was at
Missing value 3514, previous was at
Missing value 3515, previous was at
Missing value 8594, previous was at [ Time delay ]: 0.0001 Counter value: 8593
Missing value 8595, previous was at
Missing value 8596, previous was at
Done.
```

</details>

<details>
<summary>🔘 Click to see the result of Test 3</summary>

```console
./check_log_file.sh build/yarprunlog_05_07_2021_00_18_19.log
Parsing log file...
Missing value 7, previous was at [ Time delay ]: 1e-05 Counter value:  6
Missing value 9, previous was at [ Time delay ]: 1e-05 Counter value:  8
Missing value 145, previous was at [ Time delay ]: 1e-05 Counter value:  144
Missing value 197, previous was at [ Time delay ]: 1e-05 Counter value:  196
Missing value 199, previous was at [ Time delay ]: 1e-05 Counter value:  198
Missing value 549, previous was at [ Time delay ]: 1e-05 Counter value:  548
Missing value 687, previous was at [ Time delay ]: 1e-05 Counter value:  686
Missing value 1163, previous was at [ Time delay ]: 1e-05 Counter value:  1162
Missing value 1230, previous was at [ Time delay ]: 1e-05 Counter value:  1229
Missing value 1335, previous was at [ Time delay ]: 1e-05 Counter value:  1334
Missing value 1659, previous was at [ Time delay ]: 1e-05 Counter value:  1658
Missing value 1891, previous was at [ Time delay ]: 1e-05 Counter value:  1890
Missing value 2071, previous was at [ Time delay ]: 1e-05 Counter value:  2070
Missing value 2206, previous was at [ Time delay ]: 1e-05 Counter value:  2205
[... 66 lines cut ...]
When we call the 2 or more `yInfo` functions that are near in time, we expect to see their output as many times as they have been called. Unfortunately, if we compare the console output with that shown by the `yarplogger`, we can observe that in the latter case some calls are missing.

**Screenshots**

In the following screenshot we can see that `yarplogger` missed to capture the `yInfo` with counter value 298.

[image]

**Configuration (please complete the following information):**
 - OS: Windows 10 + WLS2
 - yarp version: 3.4.5 (installed with [robotology-superbuild](https://github.com/robotology/robotology-superbuild/tree/v2021.05))
-  compiler: GNU Make 4.2.1

**Additional context**

While I was providing a solution for the following assignments:

- [`assignment_yarp-find-rgb`](https://github.com/vvv-school/assignment_yarp-find-rgb/generate)
- [`assignment_make-it-roll`](https://github.com/vvv-school/assignment_make-it-roll/generate)

I was trying to debug the application using the `yInfo` functions to inspect the coverage of my code. Putting the `yInfo` calls before and after some instructions or code blocks, and looking at `yarplogger`, it seemed that some code blocks were never executed. However, after I had tried to run the code using only the console, I realized that the `yarplogger` was not showing all the information as it should have done.

As suggested by @pattacini and @traversaro, we did the test also in the configuration where we employ the `yarprun` as a broker in place of the localhost.

In this case, the results seem better (less traces missed), but the problem persists as the delay time decreases. I also noticed the `yWarning` and `yError` seem more rugged and reliable than the `yInfo`; maybe, this is due to a possible priority mechanism, I suppose.

drdanz, 2021-07-07:
I suppose this might be caused by the fact that the logger is using a BufferedPort...

https://github.com/robotology/yarp/blob/97723811b5e89806feed21a9892cc19242f27251/src/libYARP_logger/src/yarp/logger/YarpLogger.h#L220

drdanz, 2021-07-07:
CC @randaz81

PeterBowman, 2021-07-07:
> I suppose this might be caused by the fact that the logger is using a BufferedPort...

I was thinking the same, but this buffered port is configured as strict (on read):

https://github.com/robotology/yarp/blob/efc07a766fe70a293e391a4f95411bd564a2c4d8/src/libYARP_logger/src/yarp/logger/YarpLogger.cpp#L521

So perhaps the forwarding port (on write) is to be blamed, not the yarplogger's one. The log forwarder enables background writes:

https://github.com/robotology/yarp/blob/efc07a766fe70a293e391a4f95411bd564a2c4d8/src/libYARP_os/src/yarp/os/impl/LogForwarder.cpp#L43

According to the following inline doc, it does not wait for writes to complete immediately (`false`), hence I think some of them may be lost in the process:

https://github.com/robotology/yarp/blob/5313b2defd8be7237567ea56eca8dde93dfc8121/src/libYARP_os/src/yarp/os/impl/PortCore.h#L542

pattacini, 2021-07-16:
Any comment on this @randaz81 ?

drdanz, 2021-07-27:
I believe that @PeterBowman is right, the problem is probably this static bottle:

https://github.com/robotology/yarp/blob/efc07a766fe70a293e391a4f95411bd564a2c4d8/src/libYARP_os/src/yarp/os/impl/LogForwarder.cpp#L49-L59

When the `forward` method is called twice, the `Bottle::clear()` method could be called before the message is sent, therefore, if I understand correctly, the empty bottle contains no data and, when serialized as binary (not as string), nothing is sent (which is probably a different bug to be further investigated.

I opened #2667 which fixes the issue for me (at least according to @sgiraz very useful `check_log_file.sh` script). Does anyone else want to try it before I merge it?

drdanz, 2021-08-02:
>  if I understand correctly, the empty bottle contains no data and, when serialized as binary (not as string), nothing is sent (which is probably a different bug to be further investigated.

Just for reference, I was wrong on this, empty Bottles are sent correctly both in binary and text form

randaz81, 2025-08-20:
Fixed by https://github.com/robotology/yarp/pull/2789

closed 2025-08-20 by randaz81