robotology/yarp #2654: `yarp disconnect` does not work when using `mjpeg` carrier

xEnVrE, 2021-07-10:
**Describe the bug**
If a connection between two ports using `mjpeg` carries is established, then it is not possible to disconnect them. The following message error is thrown instead `[ERROR] |yarp.os.Network| Failure: no such connection`.

**To Reproduce**
Assume `/out:o` is a source of images that needs to be connected to `/in:i`. 
After connecting them with `yarp connect /out:o /in:i mjpeg`, try to disconnect them using `yarp disconnect /out:o /in:i`.

**Expected behavior**
The connection between the two ports is closed. Instead, `[ERROR] |yarp.os.Network| Failure: no such connection` is returned.

**Configuration (please complete the following information):**
 - OS: `Arch Linux`
 - yarp version: https://github.com/robotology/yarp/commit/ce0d0f2e1434e40ff3aba9265b16f5a71916b4d4
-  compiler: `gcc (GCC) 11.1.0`

GiulioRomualdi, 2021-07-10:
cc @S-Dafarra

drdanz, 2021-07-12:
Duplicate of #793

closed 2021-07-12 by drdanz