robotology/yarp #2676: Use `mkdir_p` in lieu of `mkdir` within `getHomeContextPath`

pattacini, 2021-08-05:
**Describe the bug**
Unable to create the path to `getHomeContextPath`.

Have a look at:
https://github.com/robotology/yarp/blob/0c5873cdf02a6cb4bea0d375049399dd0f556a90/src/libYARP_os/src/yarp/os/ResourceFinder.cpp#L686-L694

Because of

https://github.com/robotology/yarp/blob/0c5873cdf02a6cb4bea0d375049399dd0f556a90/doc/release/v3_5_0.md#L127-L129

I think that the function `yarpdatahome` no longer creates the desired path and we use the fallback `mkdir`.
However, the correct function for creating the directory should be `mkdir_p` instead.

**To Reproduce**
Steps to reproduce the behavior:

This [snippet](https://github.com/robotology/icub-main/blob/master/src/tools/depth2kin/src/module.cpp#L1182-L1188) should work out also in case `~/.local/share/yarp/contexts` dir is not available.

Anyway, I discovered that in one of the latest YARP versions the path doesn't get created. 

**Expected behavior**
When calling `getHomeContextPath`, we should be able then to open a file therein.

**Configuration (please complete the following information):**
 - OS: Linux
 - yarp version: `3.4.101~dev`

pattacini, 2021-08-05:
cc @traversaro

drdanz, 2021-08-18:
Fixed in #2678

PR #2678 (2021-08-18): Use `mkdir_p` instead of `mkdir` w/ `getHomeContextPath` and `getHomeRobotPath`
This PR fixes #2676.

I've tested it successfully ✔️ 

I'm proposing this change back to yarp-3.4 to then easily port it forward as for Distro 2021.08 we would likely need to rely on a fixed `yarp-3.4.6`.

I've tried to fill in the mandatory fields for describing the fix by taking inspiration from supporting files done for other old mods.
Tell me if I'm required to do anything else.
commit 3ce121ca (2021-08-18): fix #2676
closed 2021-08-18 by drdanz