robotology/yarp #2705: yarpserver stuck on Windows Command prompt

traversaro, 2021-09-03:
**Describe the bug**
I do not have a precise way of reproduce this, but lately on Windows (19042.1202) I am experiencing the following problem:
* I launch the yarpserver in a command prompt
* Then I launch something that open port (for example, a iCub simulation in Gazebo)
* The spawn of the model hangs, as the yarpserver does not reply 
* If I go to the command prompt in which I launched the yarpserver and I type a button, yarpserver is unblocked and everything starts working again

I still need to investigate this, but it seems like yarpserver goes in "hibernation" or some kind of similar state. 

**To Reproduce**
I am still work in a way to reproduce the problem.

**Expected behavior**
I would expect the yarpserver never to get stuck.

**Screenshots**
Still working on it.

**Configuration (please complete the following information):**
 - OS: Windows 10 19042.1202
 - yarp version: v3.4.6
-  compiler: Conda binaries, so no compiler was used.

**Additional context**
I want to try to reproduce this with YARP 3.5 between trying to understand better the problem.

drdanz, 2021-09-08:
Are you sure this is not the windows cmd prompt that is frozen? IIRC, when you select some text with the mouse, the windows cmd freezes the running application...

traversaro, 2021-09-08:
> 
> 
> Are you sure this is not the windows cmd prompt that is frozen? IIRC, when you select some text with the mouse, the windows cmd freezes the running application...

The Gazebo simulation hangs while waiting for the YARP ports to be registered (this can be seen in the Gazebo windows as the model is not spawned and the simulation is not going forward), so I think that also the other functionalities of the yarpserver (not only the print on the command prompt) are affected.

drdanz, 2021-09-09:
If that's the issue, the whole application is frozen, not just the print...

drdanz, 2021-09-09:
See also https://stackoverflow.com/questions/33883530/why-is-my-command-prompt-freezing-on-windows-10

traversaro, 2021-09-09:
> 
> 
> If that's the issue, the whole application is frozen, not just the print...

Exactly!

drdanz, 2021-09-10:
Sorry, that was probably not a good explanation... I mean that if you select some text on a windows cmd, the application is frozen until you unselect the text, which would explain the behaviour that you are seeing

traversaro, 2021-09-10:
> 
> 
> Sorry, that was probably not a good explanation... I mean that if you select some text on a windows cmd, the application is frozen until you unselect the text, which would explain the behaviour that you are seeing

I am not sure I was selecting anything, but I will double check again.

traversaro, 2021-10-16:
The explanation provided by @drdanz is probably the correct one, let's close the issue, I can re-open it if I have more data.

closed 2021-10-16 by traversaro