robotology/yarp #2710: Python Bindings: failing to load DLL

andrearosasco, 2021-09-08:
**Describe the bug**
When I try to import yarp from a python script I get the error 
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\robotology\robotology\share\yarp\bindings\build\lib\python3\yarp.py", line 15, in <module>
    import _yarp
ImportError: DLL load failed while importing _yarp: The specified module could not be found.
```

**To Reproduce**
On Windows 10 with Python 3.8 as the system interpreter :
1) Install the robotology superbuild
2) Build the python bindings (using CMake and Visual Studio)
Result:

```
build/
├─ lib/
│  ├─ python3/
│  │  ├─ Debug/
│  │  │  ├─ _yarp.pdb
│  │  │  ├─ _yarp.pyd
│  │  ├─ yarp.py
├─ python/
│  ├─ Debug/
│  │  ├─ _yarp.exp
│  │  ├─ _yarp.lib

```

3) Add to PYTHONPATH and Path environment variables the following directories
```
C:\robotology\robotology\share\yarp\bindings\build\lib\python3
C:\robotology\robotology\share\yarp\bindings\build\lib\python3\Debug
C:\robotology\robotology\share\yarp\bindings\build\python\Debug
```
4) Run a script that imports yarp

**Expected behaviour**
Yarp is imported correctly

**Configuration (please complete the following information):**
 - OS: Windows 10
 - yarp version: Latest robotology superbuild
-  compiler: VS19

**Additional context**
I had already successfully built the bindings for python3.9 but then, due to compatibility issues with other packages I had to switch to python3.8. I set python3.8 as the system environment, cleared the build directory and followed the steps I described above.

traversaro, 2021-09-08:
Hi @andrearosasco, 
  how did you installed the Python are you compiling against?

At a first glance, it seems that you are trying to use the Debug version of the Python bindings. As in MSVC (differently from the default toolchain used in Linux/macOS) in general Release and Debug builds are not ABI compatible, I think that if you want to use the Debug bindings you may need to use them with a Python interpreter compiled in Debug mode as well, but I do not even know if you have that installed. Another alternative could be to try to compile the Release version of the bindings, and try to use those one. In general, I would also suggest to install the Python bindings, instead of trying to use them from the build directory, as I don't know how much that workflow has been tested.

traversaro, 2021-09-08:
Sligtly OT: there is any reason you want to use the vcpkg-based installer of the robotology-superbuild, instead of using the `yarp` conda packages provided in the `robotology` channel (see https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md#binary-installation), that also provides the Python bindings already compiled for Python 3.6, 3.7, 3.8 and 3.9 ? 

If you already have a conda distribution installed (we recommend mambaforge, see https://github.com/conda-forge/miniforge#mambaforge), to create and activate a new environment called `yarpenv` with YARP bindings installed for the specified Python version you would just need to do:
~~~
conda create -n yarpenv -c conda-forge -c robotology python=3.8 yarp
conda activate yarpenv
~~~

andrearosasco, 2021-09-08:
Hi @traversaro,
the problem might have actually been related to the debug version of the bindings but I figured I don't really need them.
I was using the vcpkg-based installer just because I actually missed the fact that there was a conda package 😅. Installing everything with conda solved the problem, thank you very much!

closed 2021-09-08 by andrearosasco

traversaro, 2021-09-08:
> I was using the vcpkg-based installer just because I actually missed the fact that there was a conda package 😅.

Ack! Indeed we started documenting them in the main readme only recently, so I guess this is kind of expected. To avoid problems, if you do not use it remember to uninstall the vcpkg-based installer from "Add or remove programs", as instead the global user environment could create problems in the future.

Nicogene, 2021-09-08:
OT but related, the python bindings seems to have issues under windows, see #2640

traversaro, 2021-09-08:
Another warning is that for now we only have YARP 3.4.6 in the robotology channel, so if you need YARP 3.5 we need to wait for the next generation of conda binaries, that happens weekly on Monday night (see https://github.com/robotology/robotology-superbuild/blob/v2021.08/doc/developers-faqs.md#how-often-are-conda-binary-packages-generated). However, if you need them I can manually trigger a rebuild so that yarp 3.5.0 binaries are available.

andrearosasco, 2021-09-09:
It`s ok, I'm fine with yarp 3.4.6 for now. Thanks anyway!