robotology/yarp #2721: WebSocket progress

ste93, 2021-09-30:
## websocket:

### work done:

- websocket implementation inside yarp
- tried compiling yarp into javascript with emscripten
	code: 
	emcmake cmake .. -DYCM_DIR=/home/ste/ws/ycm/build -DSKIP_ACE=ON -DBUILD_SHARED_LIBS=OFF

	emmake make -j16 YARP_os

	/home/ste/programs/emsdk/upstream/emscripten/tools/webidl_binder ../Bottle.idl glue

	em++ -s WASM=0  -ferror-limit=0 -s LINKABLE=1 -s EXPORT_ALL=1 -s MODULARIZE=1 -I../src/libYARP_conf/src/ -Isrc/libYARP_conf/src/ -I../src/libYARP_os/src/ lib/libYARP_os.a lib/libYARP_priv_hmac.a ../bottle_idl.cpp -s STRICT=0 --bind -s LLD_REPORT_UNDEFINED --no-entry -o out.js -s EXPORT_NAME=pippo -s USE_PTHREADS  -s WASM_BIGINT

	the error returned is 
	
	wasm2js: /b/s/w/ir/cache/builder/emscripten-releases/binaryen/src/passes/I64ToI32Lowering.cpp:471: void wasm::I64ToI32Lowering::visitAtomicCmpxchg(wasm::AtomicCmpxchg *): Assertion `curr->type != Type::i64 && "64-bit AtomicCmpxchg not implemented"' failed.
    em++: error: '/home/ste/programs/emsdk/upstream/bin/wasm2js --emscripten lib.wasm --mvp-features --enable-threads --enable-mutable-globals --enable-bulk-memory --enable-sign-ext' failed (received SIGABRT (-6))

	I think the operation Cmpxchg is not implemented in 64 bits, also with lowering.
	
	the operation is in wasm (https://github.com/WebAssembly/binaryen) in file https://github.com/WebAssembly/binaryen/blob/main/src/passes/I64ToI32Lowering.cpp
	
	if you compile em++ with WASM=1 then you can reproduce the error with:
	wasm2js --emscripten lib.wasm --mvp-features --enable-threads --enable-mutable-globals --enable-bulk-memory --enable-sign-ext
	
	
 - a small javascript library (https://github.com/ste93/yarp-websocket-library) contains the websocket management together with a small bottle manager. 
  it is implemented only for bottle messages, other types of messages are not implemented (like images)_ but images can be   retrieved via image src="http://yarp http protocol"
  another functionality implemented is to query the server to get portnames.

traversaro, 2021-10-01:
Fyi @nunoguedelha

nunoguedelha, 2021-10-08:
Hello @ste93, I'm implementing for AMI-IIT lab a telemetry visualizer based on Node.js (https://github.com/ami-iit/yarp-openmct), which uses Yarp Javascript bindings and websocket support through https://github.com/robotology/yarp.js.

I'm happy to see that websocket support is making its way into Yarp native features. Is there a release plan for your implementation, or an ongoing PR?
Thanks.

nunoguedelha, 2021-11-11:
Hi @ste93 , sorry I didn't have time to react sooner to your message...

I'm not sure what you mean by "external server". There are two distinct things covered by [**robotology/yarp.js**](https://github.com/robotology/yarp.js):
-  **Javascript/Yarp-C++ bindings:** The module implements Javascript/Yarp bindings allowing to perform the usual operations, e.g. open/close/connect/disconnect Yarp ports, within the Javascript runtime environment **Node.js**. This sub-module does not require any server.
- **Websocket bridge:** It also implements a `browserCommunicator` sub-module in the same environment, which is a bridge interfacing websocket messages/requests (open/close/connect/disconnect/read/write) with the Yarp bindings that run the respective Yarp functions in the Node.js environment. An _express_ HTTP server is used to process the websocket requests. but such server can run on the localhost which is a pointer to the internal address of the loopback virtual network adapter (doesn't require a network connection).

My initial question was about the **Javascript/Yarp-C++ bindings** sub-module. Whatever "browser communicator" (bridge) is already implemented on YARP, it could re-use the binding functions in that sub-module.

closed 2021-11-12 by drdanz